Edzelf/Esp-radio #1: Missing alignment of buf array in handlebyte() function

martinayotte, 2016-04-24:
The buffer declaration need alignment attribute : 

https://github.com/Edzelf/Esp-radio/blob/master/Esp_radio.ino#L1270

It should be changed as follow : 

`
<   static uint8_t  buf[32] ;                            // Buffer for chunk
---
>   static __attribute__((aligned(4))) uint8_t  buf[32] ;                            // Buffer for chunk - See more at: http://www.esp8266.com/viewtopic.php?f=6&t=2688&start=8#sthash.ldKSwOEn.dpuf
`

Edzelf, 2016-04-24:
Yes, you are right.  Did not get any exceptions here, but I guess that it is a
change of 25 percent.
I wil update the code tomorrow.

Thanks for your feedback,
Ed.
--------------------------------------------------------------------------------

martinayotte, 2016-04-25:
Of course !
The fact that I've commented out the TFT probably make the bug coming up, I had 75% bad luck on my side ... :-)

Edzelf, 2016-04-25:
No need to comment out the TFT.  Software will work without it.  In fact it will
even work without the VS1053...

closed 2016-04-25 by Edzelf