Edzelf/Esp-radio #2: Why not move mp3client to AsyncTCP as well?

me-no-dev, 2016-04-26:
Since I see you already have the Async server up (yeah! :)), why not move the mp3 stream to async as well? You will not need to buffer the client and you can write to the VS1053 as data comes. Using that will also ACK the data only after you have it loaded to the VS, which will make networking a little more predictable :)
Let me know what you think, I can help if needed :)

Edzelf, 2016-04-26:
I will try that.

Edzelf, 2016-04-26:
Hello,

I 've tried it.  But is was no success.
It looks like the client misses some bytes now and then.
In de debug output I see in the original version:
D: ICY 200 OK
D: icy-notice1:<BR>This stream requires <a
href="http://www.winamp.com/">Winamp</a><BR>
D: icy-notice2:SHOUTcast Distributed Network Audio Server/Linux v1.9.8<BR>
D: icy-name:NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k

And in de async version:
D: ICY 200 OK
D: icy-notice1:<BR>This stream requires <a
href="http://www.winamp.com/">Winamp</a><BR>
D: icy-notice2:SHOUTcast Distributed Network Audio Server/Linux
v1.9.8<BR>icy-name:NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) -
128k

So at least one linefeed is missing.
After that, the separation between mp3-data and metadata is not possible.

You will find the new source below.

Greetings,
Ed.

--------------------------------------------------------------------------------
//******************************************************************************************
//*  Esp_radio -- Webradio receiver for ESP8266, 1.8 color display and VS1053
MP3 module.  *
//*  With ESP8266 running at 80 MHz, it is capable of handling up to 256 kb
bitrate.       *
//*  With ESP8266 running at 160 MHz, it is capable of handling up to 320 kb
bitrate.      *
//******************************************************************************************
// ESP8266 libraries used:
//  - ESP8266WiFi
//  - SPI
//  - Adafruit_GFX
//  - TFT_ILI9163C
//  - EEPROM
//  - ESPAsyncTCP
//  - ESPAsyncWebServer
//  - FS
//  - ArduinoOTA
// A library for the VS1053 (for ESP8266) is not available (or not easy to
find).  Therefore
// a class for this module is derived from the maniacbug library and integrated
in this sketch.
//
// See http://www.internet-radio.com for suitable stations.  Add the stations of
your choice
// to the table "hostlist" in the global data secting further on.  This will be
written to
// EEPROM and can be modified through remote access through the web server.
//
// Brief description of the program:
// First a suitable WiFi network is found and a connection is made.
// Then a connection will be made to a shoutcast server.  The server starts with
some
// info in the header in readable ascii, ending with a double CRLF, like:
//  icy-name:Classic Rock Florida - SHE Radio
//  icy-genre:Classic Rock 60s 70s 80s Oldies Miami South Florida
//  icy-url:http://www.ClassicRockFLorida.com
//  content-type:audio/mpeg
//  icy-pub:1
//  icy-metaint:32768          - Metadata after 32768 bytes of MP3-data
//  icy-br:128                 - in kb/sec
//
[... 1509 lines cut ...]
        }
      }
      if ( --metacount == 0 )                        
      {
        if ( metaindex )                              // Any info present?
        {
          metaline[metaindex] = '\0' ;
          // metaline contains artist and song name.  For example:
          // "StreamTitle='Don McLean - American Pie';StreamUrl='';"
          // Sometimes it is just other info like:
          // "StreamTitle='60s 03 05 Magic60s';StreamUrl='';"
          // Isolate the StreamTitle, remove leading and trailing quotes if present.
          showstreamtitle() ;                         // Show artist and title
if present in metadata
        }
        datacount = metaint ;                         // Reset data count
        chunkcount = 0 ;                              // Reset chunkcount
        datamode = DATA ;                             // Expecting data
      }
      break ;
  }
}

//******************************************************************************************
//

me-no-dev, 2016-04-26:
Somehow I can not see the source, but I was able to get the sketch running here and listen to some music, so I'll give it a go also. Bytes should not be missing

Edzelf, 2016-04-26:
Here is the source.
[Esp_radio_async.zip](https://github.com/Edzelf/Esp-radio/files/236884/Esp_radio_async.zip)

me-no-dev, 2016-04-26:
really interesting! I tried different stations and it's always that byte that is missing? always before icy-name... looking deeper... btw I meant to use AsyncClient instead of the Sync one, but I'll look into that too.

Edzelf, 2016-04-26:
Sorry, it was the wrong source indeed.  Her is the right one (I hope).
[Esp_radio_async.zip](https://github.com/Edzelf/Esp-radio/files/237246/Esp_radio_async.zip)

me-no-dev, 2016-04-26:
this is nice :) i finally found a problem with assimilating data. Onto fixing it...
Issue is that data is acked too early and the heap get's emptied quickly

me-no-dev, 2016-04-26:
delaying Acks fixed SyncClient (result is in latest git). What it showed though is that the ESP will not be able to handle the data asynchronously, because the rate of incoming packets and rate of consumption differ much. On another note, I totally removed the 20K buffer and the streams are playing fine (since you fill the buffer and right after that you read it, you can just skip it and use the buffers that the clients have).
Performance of SyncClient vs WiFiClient will not differ, and to be honest, in your exact case, you are better off using the regular WiFiClient to buffer the stream, because it uses the TCP layer's own packet buffers and saves a bit of transactions with the memory that way.
I saw you are trying to start MDNS. MDNS is now a global service and is started by ```ArduinoOTA.begin()``` or just ```MDNS.begin(hostname)```. to change the default hostname for MDNS you can call ```ArduinoOTA.setHostname("ESP-radio");``` before ```ArduinoOTA.begin();```

closed 2016-04-26 by me-no-dev

Edzelf, 2016-04-27:
Okay.  I will stick to the regular client.
I introduced the circular buffer for better performance on high bitrates only.
Lower bitrates will work without the ringbuffer.

danbicks, 2016-04-30:
Ed,

Awesome work, have one already knocked up and playing my fav stations, have you thought about adding the parametic EQ functions for bass and treble?

Great work mate, well done

Dans

Edzelf, 2016-04-30:
Thanks for kind words.  In the sourcecode you will see a remark on bass and treble as a "todo".  The next version will also save settings (preset and volume) over a restart.
Ed.

Verzonden vanaf Samsung-tablet.

danbicks, 2016-04-30:
Fantastic, roll on version 2 Woooo

Cheers mate

Dans

danbicks, 2016-05-02:
Hi @Edzelf ,

Do you have some sample routines to increase / decrease bass and treble parameters?

I now have added a simple 5 button push button panel reading from the analogue input pin, allowing me based on the 100 ms timer to increment / decrement preset, I would like to use the other buttons for EQ bass and treble for now.

Big thanks buddy and so impressed with the quality of the streams I have added :)

Cheers

Dans

Edzelf, 2016-05-03:
The new published version has bass/treble control.  Don't forget to upload the index.html as well.

danbicks, 2016-05-03:
Ed, you are simply a Guru, just got in now really excited to see how this has been implemented.

I was racking through the datasheet last night and came up with a superb idea, I am sure you may have thought about this. What about turning this the other way around and making an ESP-MP3-Encoder?

The board I have is the same as yours, the idea would be to reverse your operation and to stream the MP3 microphone encoded to a version of the radio code to pick up the stream and in effect create a point to point audio link. What do you see as the main complexity's involved with this?

Adafuit have implemented a ogg example recording to SD card, the initialization is pretty simple although will require a plugin for the vs1053 which is available. This would be seriously cool and allow for high quality point to point streaming.

Well done mate, keep up the awesome work.

Dans

Edzelf, 2016-05-04:
Nice idea.  Technically it is interesting.  The sender can act as an accesspoint, that makes it portable.  A complications might be the synchronization: what happens if the sender and receiver differ only slightly in bps-speed...  The coding have to be ogg, because the VS1053 cannot encode MP3.
I will do some experimenting with encoding, but personally I have no usefull application for point to point streaming.

danbicks, 2016-05-04:
Hi Ed,

I think it would be a really cool project. Thinking about operation, the
client "sender" attempts connection to the server basically the mp3 decoder
running ogg. Basic socket connection and then some form of simple
authentication, I.E chip id is sent from client, server checks validates
and then sends a command back to client to start sending the stream, maybe
this approach would allow synchronization to take place, I don't know a lot
about ogg format and will be researching this now. Would be superb if you
get a chance to tinker with the encoding and send stream side. I think this
system would only work point to point due to ESP limitation in remote
connections, but would be really cool concept.

Glad you are interested in this, what other stuff do you get up to in your
day job?

Big thanks

Dans

Edzelf, 2016-05-04:
I 'll see what I can do.  It will take some time...

danbicks, 2016-05-05:
Thanks buddy,

This will be really interesting.

Good luck, keep me posted

Dans

Edzelf, 2016-05-11:
@danbicks 
I have a first version of an Ogg-encoder ready for you.  See the zip-file.

danbicks, 2016-05-11:
Ed,

Wow so excited, well done you, had a look through the code, awesome mate. Just ordered another VS1053 board to try. I think for an interim I can use winamp to connect and test, will give this a go in the short term.

Any major issue's you encountered while putting this together? how did you test it and where you impressed with audio results?

Superb work, cheers Guru

Dans

Edzelf, 2016-05-11:
Dans,
It was not that difficult to cut and paste things together.  Major problem was (again) the fact that GPIO[0] and GPIO[1] are floating on my cheap Chinese board.  Then the conversion stops right after generating the Ogg header (699 words), see http://www.vsdsp-forum.com/phpbb/viewtopic.php?f=7&t=595&sid=b902cfa400af926.
Another problem is that sometimes the initialization of the board fails.  After one or 2 resets it starts suddenly and it will work for hours.  I will try to find out what the problem is.

I tested it with Firefox, and the quality (microphone input) was good.  I noticed a delay of 2.5 seconds.
I tried Winamp, but no success.

Have not tried the Esp-radio yet, but it should work.

Here some logging under normal operations:
D: IP = 192.168.2.13
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: Loading Ogg plugin
D: Result of load is 0x34
D: Waiting for connection...
pm open,type:2 0
D: New client connect
D: Starting conversion
D: Conversion started
GET / HTTP/1.1
Host: 192.168.2.13:8080
User-Agent: Mozilla/5.0 (Windows NT 10.0; WOW64; rv:46.0) Gecko/20100101 Firefox/46.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: nl,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Connection: keep-alive
D: 52970 bytes converted
D: 156678 bytes converted
D: 260922 bytes converted
D: 364848 bytes converted
D: 468096 bytes converted
D: 564968 bytes converted
D: 669830 bytes converted
D: 773106 bytes converted
D: 876926 bytes converted
D: 982852 bytes converted

danbicks, 2016-05-11:
Ed,

Amazing work, I have just downloaded Firefox and yes all connected up fine
same view as yours connection alive, I am using the onboard microphone but
no sound is coming through, did you use an external mic or line input?

Those chinese cheap boards are a pain, I have this knocked up on a NODEMCU,
I have changed original pins 4 and 5 and swapped to GPIO 0 and 2 as using
I2c. works perfect on ESP radio, not sure if this is causing the no sound
issue on ogg encoder?

I will keep tinkering, A plus mate

Dans

Edzelf, 2016-05-11:
Dans,I used the  microphone.  During conversion, you can monitor the sound through the headphone connection.

Verzonden vanaf Samsung-tablet.

Edzelf, 2016-05-12:
I made some changes to Esp-radio and Ogg-encoder and now the radio can receive data from the encoder.  Issue: the radio reads a bit faster than the encoder produces.  So the ringbuffer is always empty.  I replaced boths sketches to http://smallenburg.nl/Ogg-encoder.zip.  You can also find my mailaddress on the website, so you can mail me directly.

danbicks, 2016-05-14:
Ed,

Awesome work buddy, sorry for delay had to re build all data on to new laptop, my old one suddenly decided to type when no keys pressed lol. Too much programming for it Haha.

Super stuff buddy, I did try the last encoder and found that although mozilla locked on to the stream no audio would come through, I did plug in to the headphone socket and audio was their from the microphone strange. I will get some time hopefully next week to follow this up, finishing off my garden lighting system since the weather is really nice, features 5 outputs all PWM controlled by android app on my phone, I will send over some pics when up and running.

Keep up the amazing work.

Dans