Edzelf/Esp-radio #8: D: VS1053 error retry SB:0000 R1:FFFF R2:FFFF

praveenchaun, 2016-05-23:
Hi There,
I am getting some errors however code continue to run. So sound on my head phone finally. Do I need to modify the board to support MP3?

Serial monitor output attached.

Thanks,
[error.txt](https://github.com/Edzelf/Esp-radio/files/277036/error.txt)

praveenchaun, 2016-05-23:
I mean no sound on head phone. Thanks.

Edzelf, 2016-05-23:
Praveenchaun,

It looks like the connection to the VS1053 is not correct.  There shouldn't be errors after "Slow SPI,Testing VS1053 read/write registers...".  In your case I see a readback of "FFFF", so I assume that writing commands to SPI did not work.  Check the wiring.....

praveenchaun, 2016-05-24:
Thanks for your response.
I have not connected the reset wiring to VS1053, is that could be an issue? 
I was trying to reset the board manually pulling it to GND between below steps.
D: Reset VS1053...
D: End reset VS1053...
Could you please let me know what the diod values you used?

Edzelf, 2016-05-24:
There is no need for the reset wiring as long as you use power-off power-on to reset the radio.  I used 14148 diodes.  If "Testing VS1053 read/write registers" produces errors, check pins 1,2,4,5, and 7 of the VS1053 board.  I'm pretty sure that one of these pins is not (or wrong) wired.

praveenchaun, 2016-05-26:
Today I reconnected everything and noticed that it stops at the below location:

: Restored settings: volume 80, preset 1
D: 1Reset VS1053...
D: End reset VS1053...
D: inside testComm
D: VS1053 not properly installed!

Once I disconnect the DREQ Pin, proceeds to rest of the steps and fails!!

D: Fast SPI, Testing VS1053 read/write registers again...
D: VS1053 error retry SB:0003 R1:0000 R2:0000
D: VS1053 error retry SB:0006 R1:0000 R2:0000

Something wrong with my module. I am using ESP12E Firmware : 

2nd boot version : 1.4(b1)
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

ŒÂÕ�MEM CHECK FAIL!!!
ärlŽ‚rô
Ai-Thinker Technology Co. Ltd.
=====================
Sketch upload options as you suggested.

Edzelf, 2016-05-26:
Your ESP8266 works correctly according to the debug output.
After power up or reset the DREQ line must be HIGH. "VS1053 not properly installed!" will be printed if this line is LOW.  Pin 4 of the VS1084 should be on a 3 Volt level  when not connected.  If not, the board is probably damaged.  Are u sure that the module is powered by 5 Volt?

closed 2016-06-06 by Edzelf