Edzelf/Esp-radio #9: esp board, do not connect to internet..what I doing wrong?

yo2ldk, 2016-05-24:
Hi Ed,

 first, thank you for your work, I really need something like this !
I have two modules, one esp8266 like ESP-12F : 

http://g02.a.alicdn.com/kf/HTB1gXfyLVXXXXafXFXXq6xXFXXXk/221078962/HTB1gXfyLVXXXXafXFXXq6xXFXXXk.jpg 

and other like ESP-12E I think:

http://g01.a.alicdn.com/kf/HTB18APYLVXXXXaOaXXXq6xXFXXXb/221815479/HTB18APYLVXXXXaOaXXXq6xXFXXXb.jpg

 Now, i change the file name ssid.pw, to my ssid network, and text inside with my password. then I copy into folder Data, where the schetch it is, also folder ESP8266FS with jar file in it.
 When I upload file, I selected NodeMCU 0.9 board, then NodeMCU 1.0. 
not work  at all.. (firmware is uploaded ok) 
please some advice  I'm blocked  :(
(or on my mail - yo2ldk  a r r o u n d yahoo dot   c o m )
  thank you in advance

Edzelf, 2016-05-24:
The .pw file must be, together with index.html and favicon.ico in the map "data" (lower case).  Then you have to upload this map to your ESP8266 as shown at page 2 of the the pdf-file.  There should not be a "jar"-file in it.  Try to capture the debug output and publish it here, than I can see what's wrong.

yo2ldk, 2016-05-24:
thank you for you time and fast reply!

 please tell me also in Arduino software/tools/debbug level, what I need to select from there ?
 and, because I'm still novice, how can I read.obtain the debug file?

Edzelf, 2016-05-24:
No settings are required for debug level output.  Just start serial monitor (Ctrl+Shift+M in IDE) and copy the output.

yo2ldk, 2016-05-24:
still the same, nothing in headset, looks like is not connected via wifi , LED is on for a bit when insert power, then is off, is it was traffic, I presume it was blink, 
wires is checked and re check, firmware is ok, on serial terminal monitor is blank, nothing. 
I put out display, still nothing.
I will try later, at home router. Thank you for your patience

Edzelf, 2016-05-24:
I do not know what you mean by "firmware".  Did the upload of the sketch and the "ESP8266 Data Upload" succeed?  In the debug output you should see at least some output lines concerning WiFi connection and VS1053 tests.  As an example of the debug output see the attached file.
[debug.txt](https://github.com/Edzelf/Esp-radio/files/279410/debug.txt)

MiloPS3, 2016-05-26:
Hello, I had the same problem, I have ESP12E Lolin
you must rename .pw file at "myssid.pw" and change password and ssid inside
select Flash Size: "4 M(1M SPIFFS)".
upload this map to your ESP8266 as shown at page 2 of the pdf-file(pw file +index.html+favicon.ico)
if in debug output is (my neme ssid is "PS4" and "MO")
`D: FS Total 957314, used 9287
D: /favicon.ico                     -    766
D: /index.html                      -   6563
D: /MO.pw                           -     71
D: /PS4.pw                          -     71
`
upload success done
if in next debug output is 
`D: Selected network:                       
D: Try WiFi 
`
wrong name pw file of wrong select Flash Size
right is
`D: Selected network: PS4                      
D: Try WiFi PS4
`
if not define USETFT and use  button on D8/GPIO15 , disconnect pull-up if is used at start esp,log high on D8 is Boot on SD card (esp not start)

Excuse me for my bad English

Edzelf, 2016-05-26:
Milo, You are right about the .pw file(s).  I will make that more clear in the document file.
You are also right about GPIO15 (D8), which must be low to allow the ESP to start.  I will think of a solution for that.

MiloPS3, 2016-05-26:
change LOW to HIGH   in 
oldval3 = newval ;                          // Yes, remember value
      //if ( newval == LOW )                        // Button pushed?
	  if ( newval == HIGH )                        // Button pushed?
It is not a good solution (2 button active in low and 1 button active high)
but it works

Edzelf, 2016-05-26:
Fixed it already.

closed 2016-06-06 by Edzelf

WiGeorg, 2018-03-07:
hello Ed,
I followed this discussion (and others I found) with some given solutions to connect to WiFi, but I dont have any success! My little system only open an AP under 192.168.4.1
I use a NodeMCU 1.0(ESP 12F - LoLin), my configuration is:
4M (3M SPIFFS) => 4M(1MSPIFFS) do not work too
v2 Prebuilt (MSS0536) => v1.4Prebuilt do not work too
Arduino IDE 1.85
My pw-file is named "mySSID.pw" with the content "mySSID/mypw"

Please see the debug-file here:
[My Output.txt](https://github.com/Edzelf/Esp-radio/files/1789058/My.Output.txt)
Do you have any idea what goes wrong?

Thanks!

Willi

Edzelf, 2018-03-07:
A *.pw file is no longer required for the latest version of the radio.  You have to enter the WiFi credentials in the radio.ini file like this:
```
# WiFi network credentials, may be more than on entry.  Strongest signal will be selected.
wifi_00 = TP-LINK_9ECA/PASWORDXXX

```

WiGeorg, 2018-03-07:
Hello Ed, thank you!I started without a pw-file, but without succes. I found the pw tip in a thread. After my last mail I found out that the system connect, when I configer my router to No Security. This isn't a real solution. Standard is WEP/WEP2 with AES/TPK. It seems to be a problem of cryptage.Do you have any idea?Willi