Edzelf/Esp-radio #11: ESP-01

iqbeul, 2016-08-05:
Dear Edzelf
Im newbie for esp8266, but interested with your project, i try following your pdf guide from preparing Arduino IDE to Install Library. i dont understand how to upload the sketch what is the different update through Tools/ESP8266Sketch Data Upload with Upload like usual arrow in the main panel of Arduino IDE.
So far is unsuccesfull when i try with ESP-01 and use arduino uno as ISP, is it possible to upload and test with this board ? what should i do to get esp-01 can receive upload sketch
Thank you in advance

Edzelf, 2016-08-06:
The ESP-01 is not suitable for this project.  There are not enough pins to drive the VS1053 module and the memory is too small to load the program and the data.  I suggest to use a ESP-12 or even better a nodeMCU module.  The nodeMCU has its own USB interface so you don't need the Arduino for upload.

bickyb, 2016-08-15:
First off, awesome project.   I had a few questions:

1)I was reading through the PDF and it mentions using a "VS1063 module".  Was that supposed to be VS1053 module?

2) Let's say I want to play just a standalone MP3 file residing on a web server (e.g., www.somedomain.com/mymp3file.mp3 
How would I go about doing this?  It seems your code shows how to connect to mp3 streams, but I wanted to see if single mp3 playback is doable.

3) I wanted to see if the following board would be usable:
https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/overview

instead of the ESP-12 board.

closed 2016-08-15 by Edzelf