Edzelf/Esp-radio #12: Web page not loading completely

nlovrens, 2016-08-05:
Hi,
First of all, i would like to thank you for all what you have done, great work!
But I have one issue:
When opening web page of the esp, page is not loaded completely (see attach).
Everything else works, I can listen to music, issue additional commands thru web and so on...
Can you, please help?

Thank you

[image]

Edzelf, 2016-08-06:
Check if index.html in the data directory (the files in the SPIFFS) has the right length. It should be 6563 bytes.  If you connect a serial monitor, you should see a directory listing in the debug output.
An ESP-12 should have more than enough space to hold the files.
You could use the "show source" of your browser to examine the contents (specially near the end) of the index.html file.

nlovrens, 2016-08-06:
I couldn't find the problem. File index.html is correct size and it is listed in serial output. 
I have also tried to delete most of the stuff from index.html and leave just the header and one word in the body, then page was loaded completely. After that I have tried to add more and more stuff to body and then after some point page didn't want to load anymore. It looks like if the index.html grows over certain size then problem starts.
I have also tried to downgrade Arduino ide to 1.6.8 and ESP8266 package to 2.2.0 as stated in your PDF but it didn't help. I have also tried to flash esp as 1M(3M SPIFFS) to check if memory is corrupted, but behavior was the same.

Edzelf, 2016-08-07:
I think your analysis is right.  Could you try to use a different WiFi network (ask a neighbour or a friend)?
What kind of ESP module are you using?

nlovrens, 2016-08-07:
Hi.
After lots of troubleshooting I have found the problem. Cisco VPN client that I have installed on my PC changed MTU on all my network interfaces to 1300 (default in windows is 1500). And apparently expressIFs IP stack doesn't support MTU size less than 1460. After manually reverting my MTU to 1500 web page loaded successfully. You can close the issue.
Thank you for all your support!

closed 2016-08-08 by Edzelf