Edzelf/Esp-radio #13: Playing Standalone Mp3s

bickyb, 2016-08-15:
Hi,
I wanted to see if one can go about playing non-streamed standalone mp3s.
For example, let's say I have an Mp3 residing at www.mydomain.com/mymp3file.mp3
How would I go about playing back the file using ESP-Radio?

Edzelf, 2016-08-15:
This will not work without changing the software.  I will see if I can change the software.  It will take some time though...

Edzelf, 2016-10-11:
Standalone MP3s are now supported in the 11-oct-2016 version.

NoNeedForAName123, 2016-10-20:
Thanks for the support of standalone files.
It works fine but playing files blocks the Webinterface. 
The commands are execute after play back... 

D: Command: preset with parameter 03
D: Preset set to 3
D: Song stopped correctly after 10 msec
D: New preset requested = 3
D: Preset 3 found in .ini file
D: Connect to new host myserver/test.mp3
D: Stop client
D: Slash in station
D: Connect to preset 3, host myserver on port 80, extension /test.mp3
D: Connected to server
D: Server: Apache/2.4.10 (Debian)
D: Last-Modified: Sun, 08 Jun 2014 09:02:11 GMT
D: Accept-Ranges: bytes
D: Content-Length: 9014509
D: Connection: close
D: Content-Type: audio/mpeg
D: Switch to DATA, bitrate is 0
D: onFileRequest received /config.html
D: Command: getnetworks with parameter 0
D: onFileRequest received /radio.ini
D: onFileRequest received /favicon.ico
D: No data input
D: No data input
D: Trying other station...
D: Song stopped correctly after 0 msec
D: New preset requested = 4
D: Preset 4 found in .ini file

Edzelf, 2016-10-21:
123, I tried to reproduce it, but no success...  I used "www.stephaniequinn.com/Music/Canon.mp3" as an example, that works without problems.  Maybe your mp3 has a very high bitrate?

closed 2017-01-31 by Edzelf

texynz, 2017-08-15:
Edzelf can you confirm that:
If playing a MP3 from a file (HTTP served), that you cannot stop the file being played by forcing another file to be streamed by setting another preset.

I have tried setting two different files in the preset, but if i play preset1, and lets say it is 3min long, at 2minutes along I cannot change to preset2, and force it to play preset2. I seems like the code is not looking for/of has enough time to see another "command" whilst it is streaming preset1.
 
No issues if you are using proper streams, rather than direct file streams.