Edzelf/Esp-radio #16: Feature Request - Bootstrapping to Wifi

bickyb, 2016-08-15:
It would be really cool if you could make it so that one can connect to the ESP-radio via their phone using a browser (kind of like you do a wifi router upon initial config).  From there, it would be nice if you could search for wifi spots available, select one, and provide credentials to connect to it.  
I love the current setup but right now it requires one to have to know what wifi spots they might be connecting to and to preprogram them.  It would be cool if one could press a button to put the device in learn mode so you can configure the wifi credentials on the fly.

Edzelf, 2016-08-15:
The web server is included in the Esp-radio.  In order to connect to the webserver the ESP has has to connect to the WiFi first.  Theoretically it should be possible to let the ESP act like a station and access point simultaneously, but I do not have experience with this set-up.

But if you mean to dynamically add WiFi stations while the ESP is already connected it is possible.  This feature would need some extra software.

bbx10, 2016-08-24:
[WiFiManager](https://github.com/tzapu/WiFiManager) works well for entering new SSID and password. I have used it before.

Edzelf, 2016-08-24:
bbx10,
That looks nice!  I will implement this feature in the near future.

Edzelf, 2016-10-04:
Fixed in the current version.

closed 2016-10-04 by Edzelf