Edzelf/Esp-radio #19: webserver doesnt work correct

macmattes, 2016-09-08:
hi

thanks for this project, i use your software with a nodemcu clone from china and a vs1003 board.
streaming and audio works fin, but i am unable to use the webinterface. Some time the site loads nearly full, only the presets are missed , mostly i got a broken website which starts with "hr.open ( "GET", theUrl, false ...........  and shows the html code instead the site.
And if the site is visible i am unable to use any of the buttons, because loading never ends.
i have reflashed the esp some times also with a blink sketch and different flash sizes. nothing changed.
if i stop the site loading and use only the functions like "/?uppreset=1" it changes the channel without a delay. the other functions are also working. hope you can help to find my problem with the website.
i used safari or firefox on osx, both the same

Edzelf, 2016-09-08:
Hello,
Maybe it is a problem with the vs1003 board.  Just remove the vs1003 and try the web-interface to see if it is loading.  I tried several Chinese boards without problems.  Send you debug output to ed.at.smallenburg.nl.  (.at. = @).
Ed.

Verzonden vanaf Samsung-tablet.

Edzelf, 2016-09-08:
Are you sure that javascript works on osx?

Verzonden vanaf Samsung-tablet.

Edzelf, 2016-09-08:
Could you try to use a different wifi network?  Maybe it something to do with the maximal packet size of your router.  You may also try to load a short and a long .txt file on the SPIFFS.  Your browser should be able to load this correctly (http://<IP>/file.txt or something like that).
Verzonden vanaf Samsung-tablet.

macmattes, 2016-09-10:
hi

i am searching for a solution, it looks like a problem with the asyncwebserver or spiffs, i tried now without the decoder board and different flash spiff sizes, and also with a new nodemcu clone(same model)  nothing changes, the debug log shows the commands /?list uppreset.... all available, and i can see also that the t.md file is loaded from spiffs, but the index.html will not load.
a first test with a index.html without the script block fails, also.
i have send you my debug log, and i will search further

closed 2016-09-10 by macmattes

macmattes, 2016-09-11:
hi

everything is fine with espradio, it works on my cellphone browser. thanks
courios

Edzelf, 2016-09-11:
Take a look at this: https://support.zen.co.uk/kb/Knowledgebase/Changing-the-MTU-size-in-Mac-OS-X-10.6-to-10.9

Verzonden vanaf Samsung-tablet.