Edzelf/Esp-radio #21: sound comes out in chunks

tomtobback, 2016-09-15:
Hi, thanks a lot for this excellent project. I almost got it working and maybe you can help me to troubleshoot. I hear very short chunks of music from my VS1053. The webserver test says:
Free memory is 8992, ringbuf 0, stream 0  (is this normal?)
At startup, the Serial monitor reports that the VS1053 is not properly installed, as below.
There seems to be a problem with the reset of the VS1053, if i connect the pull-down resistor to the diodes' output, it stays LOW in permanent reset and i don't get any sound; if i disconnect the pull-down, then i get those sound chunks.
Thanks fro any suggestions- Tom

D: Reset VS1053...
D: End reset VS1053...
D: VS1053 not properly installed!
D: Fast SPI, Testing VS1053 read/write registers again...
D: VS1053 error retry SB:0000 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0003 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0006 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0009 R1:FFFF R2:FFFF
D: VS1053 error retry SB:000C R1:FFFF R2:FFFF
D: VS1053 error retry SB:000F R1:FFFF R2:FFFF
D: VS1053 error retry SB:0012 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0015 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0018 R1:FFFF R2:FFFF
D: VS1053 error retry SB:001B R1:FFFF R2:FFFF
D: VS1053 error retry SB:001E R1:FFFF R2:FFFF
D: VS1053 error retry SB:0021 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0024 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0027 R1:FFFF R2:FFFF
D: VS1053 error retry SB:002A R1:FFFF R2:FFFF
D: VS1053 error retry SB:002D R1:FFFF R2:FFFF
D: VS1053 error retry SB:0030 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0033 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0036 R1:FFFF R2:FFFF
D: VS1053 error retry SB:0039 R1:FFFF R2:FFFF
D: endFillByte is FF
D: * Scan Networks *

Edzelf, 2016-09-15:
It looks like you have the diodes mounted in reverse.
For testing you could do without the reset circuit.
First step is to get rid of the error messages.  Are you sure that the GND of the VS1053 is tightly connected to the GND of the ESP?

While not playing, ringbuf 0, stream 0 is okay (no streem, nothing buffered).
During playing the ringbuf is mostly filled up to 20000 bytes and the stream should also have some bytes in it.

tomtobback, 2016-09-16:
Hi Ed, i got rid of some errors; Slow SPI is working, but the Fast SPI test still fails as below
CPU is set to 160MHz, not sure where the speed issue comes from. the ESP-12 that i am using is quite old, maybe a hardware issue, i'll buy a new one to try. or any other ideas?
Than

D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: VS1053 error retry SB:07E9 R1:07E9 R2:0000
D: VS1053 error retry SB:13E9 R1:13E9 R2:0000
D: VS1053 error retry SB:17A9 R1:17A9 R2:0000
D: VS1053 error retry SB:1929 R1:1929 R2:0000
D: VS1053 error retry SB:1AA9 R1:1AA9 R2:0000
D: VS1053 error retry SB:1C29 R1:1C29 R2:0000
D: VS1053 error retry SB:1DA9 R1:1DA9 R2:0000
D: VS1053 error retry SB:2FA9 R1:2FA9 R2:0000
D: VS1053 error retry SB:3009 R1:3009 R2:0000
D: VS1053 error retry SB:3069 R1:3069 R2:0000
D: VS1053 error retry SB:37E9 R1:37E9 R2:0000
D: VS1053 error retry SB:38A9 R1:38A9 R2:0000
D: VS1053 error retry SB:3A29 R1:3A29 R2:0000
D: VS1053 error retry SB:3BA9 R1:3BA9 R2:0000
D: VS1053 error retry SB:3EA9 R1:3EA9 R2:0000
D: VS1053 error retry SB:43E9 R1:43E9 R2:0000
D: VS1053 error retry SB:47A9 R1:47A9 R2:0000
D: VS1053 error retry SB:4929 R1:4929 R2:0000
D: VS1053 error retry SB:4AA9 R1:4AA9 R2:0000
D: VS1053 error retry SB:4C29 R1:4C29 R2:0000
D: endFillByte is 0

Edzelf, 2016-09-16:
Looks like an SPI problem.
Did you check the ground connections?
You may try to run the ESP on 80 MHz, change line 1382 to:
system_update_cpu_freq ( 80 ) ; // Set to 80/160 MHz.
You may also try to disconnect the TFT display, it shares the SPI bus and maybe there is something wrong there.

closed 2016-10-04 by Edzelf

tomtobback, 2016-10-06:
Hi Ed,
It seems i have identified the cause of the problem: the VS1053 i am using is an Arduino shield, with a SD card, and when an SD card is inserted, it interferes with the SPI. As soon as i unplug the SD card, it works without a glitch! I tried pulling the chip select of the SD card high but that does not solve it. Maybe you have any ideas?
Thanks a lot for this great project.

Edzelf, 2016-10-06:
Yes, that makes sense.  Maybe you can find out what GPIO is used for the SD card and then redefine the pins used in the project.
Have you noticed the new version of Esp_radio (with MQTT and new configuration setup)?

tomtobback, 2016-10-07:
hi Ed, all working fine now with the new version, without display, one button on GPIO0 for next station. It seems the next station function does not check for the number of defined stations so it just keeps going up, trying even when there's no presets - is this easy to fix?
Another note; the code as GPIO0 as INPUT_PULLUP so the 10K resistor in the schematics is not necessary, right?
And in the VS1053.begin, the test for slow SPI says 2MHz in the comments but the number is 200,000 so that is rather 200kHz, right?

Edzelf, 2016-10-07:
tomtobback:
1 - Will return to preset 0 now after highest preset.
2 - Pull-up not necessary (but doesn't harm).
3- Slow SPI test is at 200 kHz indeed.

Current version has some loose ends, as I am trying to make uploads more stable and I am busy to support stand-alone mp3 files.