Edzelf/Esp-radio #22: Include support for stations with long urls.

NoNeedForAName123, 2016-10-02:
Please include support for long urls like:
http://1live.akacast.akamaistream.net/7/706/119434/v1/gnl.akacast.akamaistream.net/1live
http://kiraka.akacast.akamaistream.net/7/285/119443/v1/gnl.akacast.akamaistream.net/kiraka
http://opml.radiotime.com/Tune.ashx?id=s8254&formats=mp3

Edzelf, 2016-10-03:
Th next release of Esp-radio will have the presets in a ".ini" file instead of EEPROM and will allow long lines.  For now you may change the following:
Change line 165 to:
#define EENUM 32

Change line 166 to
#define EESIZ 128

Change line 236 to:
 "Stations from remote control NEW", // Reserved entry

Be sure that the number of station in "hostlist" is less than 32.

NoNeedForAName123, 2016-10-03:
Thx, these defines I already changed and wonder why the debug printout already shows a to short url... 
Now I changed line 170 to:
`#define DEBUG_BUFFER_SIZE 128`

But my mistake / problem was the missing (default) Port in the URL which I now correct manually:
1live.akacast.akamaistream.net**:80**/7/706/119434/v1/gnl.akacast.akamaistream.net/1live

**Problems may come if the stream is redirected like:**
http://opml.radiotime.com:80/Tune.ashx?id=s8254&formats=mp3 --> http://am.audio1.fr.ipercast.net/deluxemusic.tv/radio_web/mp3

D: Song stopped correctly after 10 msec
D: Connect to new host
D: Stop client
D: EEprom entry is opml.radiotime.com:80/Tune.ashx?id=s8254&formats=mp3
D: Slash in station
D: Connect to preset 9, host opml.radiotime.com on port 80, extension /Tune.ashx?id=s8254&formats=mp3
D: Connected to server
D: HTTP/1.1 200 OK
D: Cache-Control: private
D: Content-Type: audio/x-mpegurl; charset=utf-8
D: Server: Microsoft-IIS/8.5
D: X-AspNet-Version: 4.0.30319
D: Date: Mon, 03 Oct 2016 07:49:01 GMT
D: Content-Length: 62
D: Connection: close
D: 
D: **http://am.audio1.fr.ipercast.net/deluxemusic.tv/radio_web/mp3**
D: No data input
D: No data input
D: Trying other station...

**or another example for redirect:**

D: Command: station with parameter 2.18.214.238:80/7/706/119434/v1/gnl.akacast.akamaistream.net/1live?akacasthops=1 converted to 2
D: Song stopped correctly after 10 msec
D: Connect to new host
D: Stop client
D: EEprom entry is 2.18.214.238:80/7/706/119434/v1/gnl.akacast.akamaistream.net/1live?akacasthops=1
D: Slash in station
D: Connect to preset 11, host 2.18.214.238 on port 80, extension /7/706/119434/v1/gnl.akacast.akamaistream.net/1live?akacasthops=1
D: Connected to server
D: HTTP/1.0 302 Found
D: Location: http://2.18.214.244/7/706/119434/v1/gnl.akacast.akamaistream.net/1live?akacasthops=2
D: 
D: No data input
D: No data input
D: Trying other station...

**With the other radio station  I get some Metainformation but the stream is only printed to serial port:** 

D: Song stopped correctly after 10 msec
D: Connect to new host
D: Stop client
D: EEprom entry is 1live.akacast.akamaistream.net:80/7/706/119434/v1/gnl.akacast.akamaistream.net/1live
D: Slash in station
D: Connect to preset 8, host 1live.akacast.akamaistream.net on port 80, extension /7/706/119434/v1/gnl.akacast.akamaistream.net/1l
D: Connected to server
D: HTTP/1.0 200 OK
D: Content-Type: audio/mpeg
D: icy-description:Westdeutscher Rundfunk Koeln
D: icy-genre:WDR
D: icy-name:1Live, Westdeutscher Rundfunk Koeln
D: icy-pub:1
D: icy-url:http://www.1live.de
D: Server: Akacast 1.0
D: Cache-Control: no-cache
D: icy-metaint:16000
D: 
D: +#n
D: -uXBG`uaY=SF>zrt|"?n-4	)ET=tt7uGhR|OKA8&G!4-1>%J-`eV},~LQ/	~udw'aO>%S=N?o=:)ft&X><zA &%>nB3zX
D: o	"	l96'J]Zs(*	DIxF
D: naY}Gz`j[H\Gz%&]
D: qa{lsr,I4JZU n:	}U&80E( {,H6{xG3xAL\\d1Bz_$AD!^

Edzelf, 2016-10-04:
The new version that allows longer station URLs is ready!

closed 2016-10-04 by Edzelf