Edzelf/Esp-radio #23: Connect to the radio web server then esp reset (new version)

pasagame, 2016-10-04:
The new version is workin proberly, but if connect to radio webserver then esp8266 is be restart.
Serial Monitor show below
`Exception (28):
epc1=0x4000bf70 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffff8f0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffa90:  00000002 00000001 3ffffaf0 40214dee  
3ffffaa0:  00000002 00000000 3fff8f0c 0000473d  
3ffffab0:  00000002 00000001 3fff8f0c 4020950c  
3ffffac0:  00000000 00000000 00000000 3fff0aa0  
3ffffad0:  00000010 3ffffb60 3ffea0a0 00000000  
3ffffae0:  000003e8 3ffffb98 00000000 00000000  
3ffffaf0:  3fff90dc 0000000f 0000000a 3fff9154  
3ffffb00:  0000000f 00000001 3fff8d14 0000000f  
3ffffb10:  00000004 00000003 3ffffb60 00000018  
3ffffb20:  00000000 00000002 3ffffb80 40214d1c  
3ffffb30:  3ffe8c14 00000000 3ffffb80 00000002  
3ffffb40:  00000002 3fff8f0c 3fff8f0c 40216640  
3ffffb50:  3fff8f20 3fff90f4 3ffffb80 4020e88d  
3ffffb60:  3ffe8c14 3fff90f4 3fff8f20 40214cbc  
3ffffb70:  3fff8f20 3fff8bb4 3fff8f20 40214f3d  
3ffffb80:  00000000 3ffeb372 3fff8f0c 4020cd58  
3ffffb90:  00000002 3ffeb372 3fff8f20 40214fa0  
3ffffba0:  3ffeb372 00000000 00000000 00000002  
3ffffbb0:  00000002 3ffeb372 3fff8f0c 4020d146  
3ffffbc0:  3fff8f20 3fff8bb4 3fff8f20 00000018  
3ffffbd0:  0000001a 3ffeb35a 3fff8f0c 4020cd8e  
3ffffbe0:  3fff8f20 3ffeb35a 3fff8f20 40214fa0  
3ffffbf0:  3ffeb35a 00000000 00000000 00000018  
3ffffc00:  0000001a 3ffeb35a 3fff8f0c 4020cf7c  
3ffffc10:  3fff8f20 3fff8bb4 3fff8f20 00000022  
3ffffc20:  0000003c 3ffeb338 3fff8f0c 4020cd8e  
3ffffc30:  3fff8f20 3ffeb338 3fff8f20 40214fa0  
3ffffc40:  3ffeb338 00000000 00000000 00000022  
3ffffc50:  0000003c 3ffeb338 3fff8f0c 4020cf7c  
3ffffc60:  3fff8f20 3fff8bb4 3fff8f20 00000020  
3ffffc70:  0000005c 3ffeb318 3fff8f0c 4020cd8e  
3ffffc80:  3fff8f20 3ffeb318 3fff8f20 40214fa0  
3ffffc90:  3ffeb318 00000000 00000000 00000020  
3ffffca0:  0000005c 3ffeb318 3fff8f0c 4020cf7c  
3ffffcb0:  3fff8f20 3fff8bb4 3fff8f20 00000042  
3ffffcc0:  0000009e 3ffeb2d6 3fff8f0c 4020cd8e  
3ffffcd0:  3fff8f20 3ffeb2d6 3fff8f20 40214fa0  
3ffffce0:  3ffeb2d6 00000000 00000000 00000042  
3ffffcf0:  0000009e 3ffeb2d6 3fff8f0c 4020cf7c  
3ffffd00:  3fff8f20 3fff8bb4 3fff8f20 00000049  
3ffffd10:  000000e7 3ffeb28d 3fff8f0c 4020cd8e  
3ffffd20:  3fff8f20 3ffeb28d 3fff8f20 40214fa0  
3ffffd30:  3ffeb28d 00000000 00000000 00000049  
3ffffd40:  000000e7 3ffeb28d 3fff8f0c 4020cf7c  
3ffffd50:  3fff8f20 00000001 3fff8f20 00000056  
3ffffd60:  0000013d 3ffeb237 3fff8f0c 4020cd8e  
3ffffd70:  3fff8f20 3ffeb237 3fff8f20 40214fa0  
3ffffd80:  3ffeb237 00000000 00000000 00000056  
3ffffd90:  0000013d 3ffeb237 3fff8f0c 4020cf7c  
3ffffda0:  3ffe8c14 3fff8f20 3fff8f20 00000017  
3ffffdb0:  00000154 3ffeb220 3fff8f0c 4020cd8e  
3ffffdc0:  3fff8f20 3ffeb220 3fff8f20 40214fa0  
3ffffdd0:  3ffeb220 00000000 00000000 00000017  
3ffffde0:  00000154 3ffeb220 3fff8f0c 4020cf7c  
3ffffdf0:  3ffe8c14 3fff8f0c 3fff8f20 00000032  
3ffffe00:  00000186 3ffeb1ee 3fff8f0c 4020cd05  
3ffffe10:  3fff8f20 3ffeb1ee 3fff8f20 40214fa0  
3ffffe20:  3ffeb1ee 3ffee350 3fff8ec8 00000032  
3ffffe30:  00000186 3ffeb1ee 3fff8f0c 4020cf7c  
3ffffe40:  3ffeb140 00000000 00000000 4010068c  
3ffffe50:  00000000 01164119 4010292a 00000100  
3ffffe60:  3fff8f20 3ffeb140 3ffeb140 00000001  
3ffffe70:  00000001 00007fff 000000f6 3fff8ea0  
3ffffe80:  00000001 3fff2b7c 3fff8e4c 4020d171  
3ffffe90:  3fffc278 401026c4 3fff8e4c 4020b31c  
3ffffea0:  3fff8264 3fff07c4 3fff8264 40235679  
3ffffeb0:  3fff23b4 3fff23c0 00000006 3fff21c8  
3ffffec0:  3fff8264 3fff21d0 3fff21cc 4020b36f  
3ffffed0:  3fff8264 3fff21d0 3fff21cc 4023adf4  
3ffffee0:  00000186 ea01a8c0 00000018 00000186  
3ffffef0:  00000018 00000000 3fff28e4 40106d84  
3fffff00:  00000000 00000000 3ffebf07 3fff23b8  
3fffff10:  3ffeb1c6 3fff23c0 3fff2b7c 402392ed  
3fffff20:  3fff20f4 3fff2874 3fff2874 3ffef470  
3fffff30:  00000000 3fff2b7c 0000001c 3fff2874  
3fffff40:  3ffeb1b8 00000000 3fff2b7c 402386e9  
3fffff50:  0901a8c0 000001e4 00000000 00000026  
3fffff60:  00000002 0000001a 40219137 3ffee350  
3fffff70:  3ffeb190 3fffdcc0 3ffea988 3ffea988  
3fffff80:  402190aa 3ffee350 00000000 3fff28ec  
3fffff90:  3fffdc80 00000000 3fff2b7c 4022d17b  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld`

Edzelf, 2016-10-04:
Did you ulpoad the data directory?
Is this the beginning of the log output?

pasagame, 2016-10-04:
yes i upload data directory.
This log does not show the beginning, But will show only when are connect to radio webserver and radio webserver not running (still loading) and esp8266 will reset. if disconnect  from radio webserver, esp8266 will return work proberly

Edzelf, 2016-10-04:
I changed one line in the code.  This had to do with wrong handling index.html page.
Please try with the new source (line 2207).

pasagame, 2016-10-04:
Now working. Thank You verry much.

closed 2016-10-04 by pasagame