Edzelf/Esp-radio #24: Include default http port 80 if not set in URL (URL redirection HTTP 302)

NoNeedForAName123, 2016-10-04:
Please add support for default http port if not set in URL (especially wit URL redirection HTTP 302) :

D: Preset 1 found in .ini file
D: Remember preset 1
D: Connect to new host kiraka.akacast.akamaistream.net:80/7/285/119443/v1/gnl.akacast.akamaistream.net
D: Stop client
D: Slash in station
D: Connect to preset 1, host kiraka.akacast.akamaistream.net on port 80, extension /7/285/119443/v1/gn
D: Connected to server
D: HTTP/1.0 302 Found
D: Location: http://195.27.154.151/7/285/119443/v1/gnl.akacast.akamaistream.net/kiraka?akacasthops=1
D: 
D: No data input

NoNeedForAName123, 2016-10-04:
sorry I was wrong way. `D: No data input` is not coming from the missing port :-( 
My stream are vbr (variable bit rate). If set the bitrate in the  "datamode == INIT" Block to 128 instead of 0 the music is played with gaps.
Line 1639 looks like:
   bitrate = 192 ;                                      // Bitrate still unknown

closed 2016-10-04 by NoNeedForAName123

Edzelf, 2016-10-05:
This stream is not a standard Icecast stream.  The "icy-headers" are missing.
The software expect something like:
.
.
D: icy-br:192
D: ice-audio-info: bitrate=192
D: icy-br:192
D: icy-genre:Talk
D: icy-name:Radio 1
D: icy-pub:0
D: icy-url:http://www.radio1.nl
D: icy-metaint:16000
.
.

NoNeedForAName123, 2016-10-05:
The stream itself sound very good. There are only a few gaps, not very often. This morning i saw on my squuezebox that the stream delivers cover/station/album art. I think that the problem. Do you have an idea how this pictures are encoded/embedded? Is there a spec available? 
I only found: http://wiki.shoutcast.com/wiki/SHOUTcast_2_%28Ultravox_2.1%29_Protocol_Details
--> Known Message Types --> 
class     type        description
....
0x4 	0x0xx 	Station logo (see 'Image Notes' for xx details)
0x4 	0x1xx 	Album art (see 'Image Notes' for xx details) 
.....
Where xx is
00 	image/jpeg
01 	image/png
02 	image/bmp
03 	image/gif