Edzelf/Esp-radio #25: Make AsyncMqttClient.h optional like USETFT

NoNeedForAName123, 2016-10-04:
please add download reference to AsyncMqttClient.h to documentation and make the usage of Mqtt configurable.

NoNeedForAName123, 2016-10-04:
pls. see https://github.com/NoNeedForAName123/Esp-radio/blob/master/Esp_radio.ino how it worked for me.

Edzelf, 2016-10-05:
If you mean "upload", thats already documented.  You can switch off Mqtt by removing the "mqtt"-lines in the radio.ini file.  I documented the last feature in the new pdf-file.

NoNeedForAName123, 2016-10-05:
Sorry, I mean in the ESP-radio.pdf in the chapter "Arduino IDE installation for Esp-radio" there is the download location for the AsyncMqttClient missing. So even with switching off mqtt in the ini file compilation fails:

`/Esp-radio/Esp-radio.ino:113:29: fatal error: AsyncMqttClient.h: No such file or directory
 #include <AsyncMqttClient.h>

https://github.com/marvinroger/async-mqtt-client
                             ^
compilation terminated.`

Edzelf, 2016-10-11:
Fixed.

closed 2016-10-11 by Edzelf