Edzelf/Esp-radio #27: Some stations will not work

RingoK, 2016-10-09:
icecast.err.ee:80/raadio2.mp3
37.0.26.55/HitFM_192

Whats the problem?

Thank You!

Edzelf, 2016-10-10:
These station do not send a standard ICY header.
At this moment I do not understand the format, but I will look into that.

Edzelf, 2016-10-11:
The problem is that these stations do not include the bitrate in the header.  Fixed that in the 11-oct version.

closed 2016-10-11 by Edzelf

RingoK, 2016-10-11:
i dont know, dont work

Edzelf, 2016-10-11:
When I enter: "icecast.err.ee:80/raadio2.mp3" in the Serial Monitor, I see in the logging:

D: Command: station with parameter icecast.err.ee:80/raadio2.mp3
D: New preset station icecast.err.ee:80/raadio2.mp3 accepted
D: Connect to new host icecast.err.ee:80/raadio2.mp3
D: Stop client
D: Slash in station
D: Connect to preset 6, host icecast.err.ee on port 80, extension /raadio2.mp3
D: Connected to server
D: Server: Icecast 2.4.2
D: Content-Type: audio/mpeg
D: Cache-Control: no-cache
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Pragma: no-cache
D: icy-name:Raadio2
D: icy-pub:0
D: icy-metadata:1
D: icy-metaint:16000
D: Switch to DATA, bitrate is 0
D: Metadata block 32 bytes
D: Streamtitle found, 23 bytes
D: StreamTitle='Raadio 2';

Please compare this with your output.  AR you sure you are using the 11-oct version:
D: Starting ESP Version 11-oct-2016...  Free memory 15160