Edzelf/Esp-radio #28: power supply

tomtobback, 2016-10-11:
Hi Ed, thanks again for this wonderful project, working well now.
Some notes that might be of interest; when i tried to power both ESP and amp (small 5V stereo class D amp) from the same 5V, i found out that the VS1053 audio output is DC biased to 1.25V, which seems to be the case on most boards. There's an application note explaining how to get rid of the bias (blocking caps etc), but that still had a lot of noise probably due to the shared power supply (or the breadboard setup). So i also opted for 2 separate power supplies, and the sound is very clean. 
I like the switch circuit in your PDF but i was wondering if this is not a little dangerous, to charge 2 lithium cells at the same time, with 1 charger which cannot know the state of both cells. Additionally, when you switch your radio off, the (presumably) fully charged batt1 is directly connected to the less than fully charged batt2, which i think can lead to high currents, potentially dangerous if your lithium cells do not each have protection circuits.
The ESP seems to need more than 100mA constantly, just out of interest, how many hours do you get out of your battery approximately?
As above is not really an issue, feel free to close it.

Edzelf, 2016-10-11:
Thank you for your feedback.  You are right.  The current may be very high.  For simplicity I think might be sufficient to add a current limiting resistor, like 5.6 ohms, to the load line of "battery 2".  An extra charger would also do the trick as well as using cells with protection circuits.
I cannot tell how long the radio runs on batteries.  Normally it is connected to a power adapter, so it will be charged when I switch of to "off".  In the "On" state, only the battery for the amplifier is discharged (26 mA).  The current for the ESP8266 plus the VS1053 is 135 mA, so it should work for about 15 hours.
Have you tried my new version?  It now can play standalone MP3s as well.

tomtobback, 2016-10-15:
Hi Ed, i was a bit confused, you mean mp3 files on another server, right?
for a moment i thought we could upload mp3 files to the ESP, at the bottom of the config screen, but i guess that's not what this upload is intended for?

Edzelf, 2016-10-15:
Esp-radio will play mp3 file that are placed on a remote server.  You can however upload any file to the ESP, it was meant to make changes to the html files.  I do have a version that is capable of playing from SPIFFS by specifying "localhost" as the server in the station, but since the SPIFFS can only store 3 MB it is not very useful.
The SPIFFS is not very stable.  I noticed that reading from a file is very slow sometimes.
However, my TFT has an on-board "MicroSD Card Breakout".  Do you think it's worthwhile to have this implemented?

tomtobback, 2016-11-16:
Hi Ed, it took me a while to finish this project, it's working very well now. to answer your question, personally i'm not too interested in playing music from SD card so wouldn't vote for that.
it's very functional for me the way it is now.
i've posted some pics on my blog if you'd like to see my build: http://cassiopeia.hk/internetradio/

Edzelf, 2017-01-31:
Finally saw your blog.
Nice box and nice story too!

closed 2017-02-22 by tomtobback