Edzelf/Esp-radio #29: Problem with library AsyncMqttClient

Lubosno, 2016-10-14:
Thank you for super project.
New version i can not compile.
I have error  in line 1566  mqttclient.onConnect ( onMqttConnect ) ;
Please advice and help
Lubosno from Czech rep.

This is my error report:
> Arduino: 1.6.8 (Windows XP), Vývojová deska: "Generic ESP8266 Module, 160 MHz, 80MHz, DIO, >115200, 4M (1M SPIFFS), ck, Disabled, None"

> C:\Documents and Settings\sklad\Dokumenty\Arduino\Esp-radio-master (1)\Esp-radio-master\Esp->radio-master.ino: In function 'void setup()':

>Esp-radio-master:1566: error: no matching function for call to 'AsyncMqttClient::onConnect(void (&)())'
>       mqttclient.onConnect ( onMqttConnect ) ;

>C:\Documents and Settings\sklad\Dokumenty\Arduino\Esp-radio-master (1)\Esp-radio-master\Esp->radio-master.ino:1566:44: note: candidate is:

>In file included from C:\Documents and Settings\sklad\Dokumenty\Arduino\libraries\async-mqtt-client->master\src/AsyncMqttClient.h:4:0,

> from C:\Documents and Settings\sklad\Dokumenty\Arduino\Esp-radio-master (1)\Esp-radio->master\Esp-radio-master.ino:116:

>C:\Documents and Settings\sklad\Dokumenty\Arduino\libraries\async-mqtt-client->master\src/AsyncMqttClient.hpp:43:20: note: AsyncMqttClient& >AsyncMqttClient::onConnect(AsyncMqttClientInternals::OnConnectUserCallback)
 >  AsyncMqttClient& onConnect(AsyncMqttClientInternals::OnConnectUserCallback callback);
>C:\Documents and Settings\sklad\Dokumenty\Arduino\libraries\async-mqtt-client->master\src/AsyncMqttClient.hpp:43:20: note:   no known conversion for argument 1 from 'void()' to >'AsyncMqttClientInternals::OnConnectUserCallback {aka std::function<void(bool)>}'
>exit status 1
>no matching function for call to 'AsyncMqttClient::onConnect(void (&)())'

Edzelf, 2016-10-14:
Looks like you have the wrong library for MQTT.  This project uses "AsyncMqttClient by Marvin Roger Version 0.3.0".  See https://github.com/marvinroger/async-mqtt-client.

This is a list of the libraries used (at the end of the compiler logging:
ESP8266WiFi op versie 1.0 in map: \AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi
ESPAsyncTCP-master op versie 1.0.0 in map: \Arduino\libraries\ESPAsyncTCP-master
ESPAsyncWebServer-master op versie 1.0.0 in map: \Arduino\libraries\ESPAsyncWebServer-master
async-mqtt-client-master op versie 0.3.0 in map: \Arduino\libraries\async-mqtt-client-master
SPI op versie 1.0 in map: \AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI
Adafruit_GFX_Library op versie 1.1.5 in map: \Arduino\libraries\Adafruit_GFX_Library
TFT_ILI9163C-master op versie 0.9 in map: \Arduino\libraries\TFT_ILI9163C-master
Ticker op versie 1.0 in map: \Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Ticker
ArduinoOTA op versie 1.0 in map: \Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ArduinoOTA
Hash op versie 1.0 in map: \Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Hash
ESP8266mDNS in map: \Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS (legacy)

Lubosno, 2016-10-14:
Hallo, i have lastes version from internet 0.5.0

This is my log

> async-mqtt-client-master ve verzi 0.5.0 v adresáři: C:\Documents and Settings\sklad\Dokumenty\Arduino\libraries\async-mqtt-client-master 

the others library are the same

Thank

Edzelf, 2016-10-14:
I updated the code for using 0.5.0 version of the library.

closed 2016-10-14 by Edzelf