Edzelf/Esp-radio #31: function "muteflag" not working  and Configuration file cannot be edited through web interface.

pasagame, 2016-10-22:
1. function "muteflag" not working, i test by MQTT (mute, unmute) no effect.
2. Configuration file can not be edited or upload through web interface. if have edit or uploading files "Radio.ini"  from the Web server some data on file is lost and espradio cannot working. Be solved by upload via "ESP8266 Sketch Data Upload" on arduino ide only then will work properly.

http://www.mx7.com/view2/zr9d6GpCasnkljmj

Edzelf, 2016-10-22:
1. mute/unmute works now.
2. Sometimes uploads are damaged.  It looks like there is a problem in the SPIFFS library.

pasagame, 2016-10-23:
now i test function mute/unmute not work.

Edzelf, 2016-10-23:
Did you rebuild the project?
Check line 111 in your sketch.  It should read "#define VERSION "22-oct-2016".
I tested it in the webinterface:
http://192.168.2.12/?mute
http://192.168.2.12/?unmute

pasagame, 2016-10-23:
thank you, i update to lastest version. mute function working proberly.

closed 2016-10-23 by Edzelf