Edzelf/Esp-radio #32: Reset connection in schematic different from wiring diagram

harmverbeek, 2016-10-22:
In the schematic the RST pin of the ESP12 seems to be connected to +5V(!) and the RST pin of the LCD.
In the wiring table it is supposed to connect to the RST pins of both LCD and VS1053.

What is the right way?

Edzelf, 2016-10-22:
Harm,
The drawing was not correct.  I 've uploaded a new one.  The reset of the ESP is pulled high by a resistor on the mounting board.  The reset of the TFT and the VS1053 is done by the reset circuit (2 diodes and a resistor): when GPIO5 and GPIO16 are both low, the reset pins go low.

68erDirk, 2016-11-13:
What is the purpose of the reset circuit? I left it away completely and only connected everything as described under "wiring"? Everything seems to work well.

snazzware, 2017-03-21:
FYI, I recently loaded up this project on a Wemos D1 mini, and just wired the Reset pin of the VS1053 board to the D1's reset pin. Worked like a charm!

Edzelf, 2017-03-24:
snazzware: thats fine.  But this pin is not programmable, so the VS1053 will not be reset on powerup.  This reset, however, seems to be not necessary in practice.  So all will work correctly, even if you connect the reset pin of the VS1053 to +5 Volt.