Edzelf/Esp-radio #33: TFT display not working

pasagame, 2016-10-27:
I use a 1.8 TFT, but it does not work. I used standalone. (Test code), works properly. please look picture this link http://www.mx7.com/view2/zrLdZLyC7TSCjbPc

standalone (test code)

#include <Adafruit_GFX.h>
#include <TFT_ILI9163C.h>
#include <SPI.h>

// Color definitions for the TFT screen (if used)
#define  BLACK   0x0000
#define BLUE    0xF800
#define RED     0x001F
#define GREEN   0x07E0
#define CYAN    GREEN | BLUE
#define MAGENTA RED | BLUE
#define YELLOW  RED | GREEN
#define WHITE   0xFFFF

#define TFT_CS 15
#define TFT_DC 2

TFT_ILI9163C     tft = TFT_ILI9163C ( TFT_CS, TFT_DC ) ;

void setup() {
  SPI.begin() ;
  tft.begin() ;                                      // Init TFT interface
  tft.fillRect ( 0, 0, 160, 128, BLACK ) ;           // Clear screen does not work when rotated
  tft.setRotation ( 3 ) ;                            // Use landscape format
  tft.clearScreen() ;                                // Clear screen
  tft.setTextSize ( 1 ) ;                            // Small character font
  tft.setTextColor ( WHITE ) ;  
  tft.println ( "Starting" ) ;
  
}

void loop(){
//
}

Edzelf, 2016-10-27:
At the moment I have no possibility to look at the code.  If it works stand-alone, try to disconnect the VS1053 and the resetlines.
Verzonden vanaf Samsung-tablet.

#include 

#include

closed 2017-01-31 by Edzelf

pasagame, 2017-09-17:
until now i cannot use tft display.