Edzelf/Esp-radio #36: ESP Record voice by vs1053

etoshey, 2016-11-02:
Hello,
I wiered an esp wroom and vs1053 mudule (geeethec) , then use your esp radio code (that is great and usefull source code) , it works fine.
You write a comment on your code about defualt midi mode , i have to change for record voice?
I am trying to record voice  but i have problem . Dreq pin gone low when start to record.
At the first, in my code :
cs pin ==> HIGH
dcs pin==> HIGH
dreq pin define as an INPUT
SPISettings => 250000, MSBFIRST, SPI_MODE0 
and then soft reset (SCI_MODE= SM_SDINEW | SM_RESET) 
Now, SCI_STATUS register equals 72 (0010 1010). Sine Test is OK, 
record initiation :
SCI_CLOCKF = 0xC000 // set max clock
wait for rising DREQ pin
SCI_BASS = 0 // clear Bass 
SCI_MODE= SM_SDINEW | SM_RESET // soft reset
wait for rising DREQ pin
SCI_WRAMADDR =0xC01A
SCI_WRAM = 0x02
SCI_AICTRL0= 1024
SCI_AICTRL1= 1024
SCI_AICTRL2= 0
SCI_AICTRL3= 0;
loadPlugin...
Active 
SCI_MODE = SM_ADPCM | SM_SDINEW
SCI_AIADDR = 0x34 
wait for rising DREQ pin ==> INFINITY LOOP, because the DREQ pin stay down.

I can not find my mistake , If is possible guide me, Thanks

Edzelf, 2016-11-15:
I did some experimenting with recording (ogg-format).  Not with the best results....  You can examine my code on http://smallenburg.nl/Ogg-encoder.zip.
Note that you have to supply a file for your WiFi SSID containing the password, for example SSID.pw.

closed 2016-11-15 by Edzelf

fredericplante, 2016-12-13:
At line 18, you specifies "Pin to program". What does that mean? What is hooked on those pins?

Edzelf, 2016-12-14:
It is the pinnumber that you use in the sketch to address the pin.  Note the issue I mentioned at the top of the Ogg-encoder sketch: "Sometimes the conversion will not start"....
GPIO[0] and GPIO[1] are being set to LOW in the sketch.

fredericplante, 2016-12-14:
So it basically, it's the same pin out then the Radio-player, except for Xrst that is set to pin 3, instead of the reset pin of the ESP8266.

Edzelf, 2016-12-14:
Could be, this is a different project, so the pin configuration and software will differ.

fredericplante, 2016-12-21:
Before I buy another VS1053, tel me will this broadcaster project, the zip file you just sent, and the radio project connect and play along together?

Edzelf, 2016-12-21:
Yes, I had it working for a while.  But it was not very reliable.  I think it will be a challenge to get it working.

fredericplante, 2016-12-21:
Well ok then, it will be part of my late xMass shopping. Thank you Ed