Edzelf/Esp-radio #37: Panic & reboot

jammoyano, 2016-11-08:
It's unable to play or use the radio .... just reboot after reboot.

Here is the serial monitor ... ¿any idea?

`D: /config.html                     -   3817
D: /favicon.ico                     -    766
D: /index.html                      -   6769
D: /radio.ini                       -    538
D: Added SSID 00 = Wlan_99d5_2 to acceptable networks
D: * Scan Networks *
D: Number of available networks: 7
D:  1 - ONO_2016                  Signal: -87 dBm Encryption Auto  
D:  2 - _AUTO_ONOWiFi             Signal: -86 dBm Encryption None  
D:  3 - Orange-5F28               Signal: -89 dBm Encryption Auto  
D:  4 - Ptv Maria Jose            Signal: -91 dBm Encryption Auto  
D:  5 - MOVISTAR_B888             Signal: -93 dBm Encryption WPA   
D:  6 - Wlan_99d5_2               Signal: -82 dBm Encryption WPA2  Acceptable
D:  7 - Ptv_4729_2                Signal: -83 dBm Encryption WPA2  
D: --------------------------------------
D: Command: mqttbroker with parameter broker.hivemq.com
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter none
D: Command: mqtttopic with parameter espradio
D: Command: mqttpubtopic with parameter espradioIP
D: Command: wifi_00 with parameter Wlan_99d5_2/xxxxxxx
D: Starting ESP Version 22-oct-2016...  Free memory 16880
D: Sketch size 321056, free size 724992
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Selected network: Wlan_99d5_2              
D: Try WiFi Wlan_99d5_2
D: IP = 10.0.0.70
D: Start server for commands
D: list request from browser

Panic C:\Users\Jose\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\core_esp8266_main.cpp:98 __yield

ctx: sys 
sp: 3ffff840 end: 3fffffb0 offset: 01b0

>>>stack>>>
3ffff9f0:  3fff7ac8 00000007 3ffffa40 40216b35  
3ffffa00:  0000000a 3ffffa58 3ffffa40 402153a5  
3ffffa10:  3ffe0070 0000021a 3ffffa58 4021434d  
3ffffa20:  3ffe8be0 3fff865c 3ffffa40 402143db  
3ffffa30:  00000000 3fff865c 3fff8bc4 4020858e  
3ffffa40:  3fff8ac4 0000001f 0000001e 40214139  
3ffffa50:  00000000 00000000 3ffea0d8 00000000  
3ffffa60:  000003e8 00003b60 3fff921c 3fff8b94  
3ffffa70:  3ffe90fc 3fff0ad4 3fff0298 3fff925c  
3ffffa80:  0000005f 0000005a 3fff9314 0000005f  
3ffffa90:  0000005a 3fff87ec 0000000f 0000000a  
3ffffaa0:  3fff8ac0 3fff2bcc 3fff865c 00003b56  
3ffffab0:  00000002 00000001 3fff865c 40209715  
3ffffac0:  00000000 00000000 00000000 3fff0aac  
3ffffad0:  00000010 3ffffb60 3ffea0d8 00000000  
3ffffae0:  000003e8 3ffffb98 00000000 00000000  
3ffffaf0:  00000010 3ffffb80 00000000 3fff2abc  
3ffffb00:  0000000f 00000001 3fff2bcc 0000000f  
3ffffb10:  00000004 00000003 3ffffb60 00000000  
3ffffb20:  00000000 3ffffb80 3ffffb80 4021490c  
3ffffb30:  3ffe8ca0 00000000 3ffffb80 00000002  
3ffffb40:  00000002 3fff865c 3fff865c 40216288  
3ffffb50:  3fff8670 3fff8804 3ffffb80 4020eb69  
3ffffb60:  3ffe8ca0 3fff8804 3fff8670 402148ac  
3ffffb70:  3fff8670 3fff292c 3fff8670 40214b2d  
3ffffb80:  00000000 3ffeb9c0 3fff865c 4020d000  
3ffffb90:  00000002 3ffeb9c0 3fff8670 40214b90  
3ffffba0:  3ffeb9c0 00000000 00000000 00000002  
3ffffbb0:  00000002 3ffeb9c0 3fff865c 4020d3f6  
3ffffbc0:  3fff8670 3fff292c 3fff8670 00000018  
3ffffbd0:  0000001a 3ffeb9a8 3fff865c 4020d036  
3ffffbe0:  3fff8670 3ffeb9a8 3fff8670 40214b90  
3ffffbf0:  3ffeb9a8 00000000 00000000 00000018  
3ffffc00:  0000001a 3ffeb9a8 3fff865c 4020d228  
3ffffc10:  3fff8670 3fff292c 3fff8670 0000001c  
3ffffc20:  00000036 3ffeb98c 3fff865c 4020d036  
3ffffc30:  3fff8670 3ffeb98c 3fff8670 40214b90  
3ffffc40:  3ffeb98c 00000000 00000000 0000001c  
3ffffc50:  00000036 3ffeb98c 3fff865c 4020d228  
3ffffc60:  3fff8670 3fff292c 3fff8670 00000020  
3ffffc70:  00000056 3ffeb96c 3fff865c 4020d036  
3ffffc80:  3fff8670 3ffeb96c 3fff8670 40214b90  
3ffffc90:  3ffeb96c 00000000 00000000 00000020  
3ffffca0:  00000056 3ffeb96c 3fff865c 4020d228  
3ffffcb0:  3fff8670 3fff292c 3fff8670 00000036  
3ffffcc0:  0000008c 3ffeb936 3fff865c 4020d036  
3ffffcd0:  3fff8670 3ffeb936 3fff8670 40214b90  
3ffffce0:  3ffeb936 00000000 00000000 00000036  
3ffffcf0:  0000008c 3ffeb936 3fff865c 4020d228  
3ffffd00:  3fff8670 3fff292c 3fff8670 00000049  
3ffffd10:  000000d5 3ffeb8ed 3fff865c 4020d036  
3ffffd20:  3fff8670 3ffeb8ed 3fff8670 40214b90  
3ffffd30:  3ffeb8ed 00000000 00000000 00000049  
3ffffd40:  000000d5 3ffeb8ed 3fff865c 4020d228  
3ffffd50:  3fff8670 00000001 3fff8670 00000057  
3ffffd60:  0000012c 3ffeb896 3fff865c 4020d036  
3ffffd70:  3fff8670 3ffeb896 3fff8670 40214b90  
3ffffd80:  3ffeb896 00000000 00000000 00000057  
3ffffd90:  0000012c 3ffeb896 3fff865c 4020d228  
3ffffda0:  3ffe8ca0 3fff8670 3fff8670 00000011  
3ffffdb0:  0000013d 3ffeb885 3fff865c 4020d036  
3ffffdc0:  3fff8670 3ffeb885 3fff8670 40214b90  
3ffffdd0:  3ffeb885 00000000 00000000 00000011  
3ffffde0:  0000013d 3ffeb885 3fff865c 4020d228  
3ffffdf0:  00000002 40100fab 3fff8670 00000033  
3ffffe00:  00000170 3ffeb852 3fff865c 4020cfa9  
3ffffe10:  3fff8670 3ffeb852 3fff8670 40214b90  
3ffffe20:  3ffeb852 3ffee370 3ffffe70 00000033  
3ffffe30:  00000170 3ffeb852 3fff865c 4020d228  
3ffffe40:  3ffeb16c 00000000 00000000 4010068c  
3ffffe50:  00000000 00e7c449 4010292a 00000100  
3ffffe60:  3fff8670 3ffeb16c 3ffeb16c 00000001  
3ffffe70:  00000001 00007fff 000000d6 3fff2b60  
3ffffe80:  00000001 3fff2a8c 3fff2b0c 4020d421  
3ffffe90:  3fffc278 401026c4 3fff2b0c 4020b50c  
3ffffea0:  3fff85a4 3fff07d8 3fff85a4 402350d1  
3ffffeb0:  3fff23c4 3fff23d0 00000006 3fff21d8  
3ffffec0:  3fff85a4 3fff21e0 3fff21dc 4020b55f  
3ffffed0:  3fff85a4 3fff21e0 3fff21dc 4023a84c  
3ffffee0:  00000170 4600000a 00000018 00000170  
3ffffef0:  00000018 00000000 3fff2c8c 40106d84  
3fffff00:  00000000 00000000 3ffeb84a 3fff23c8  
3fffff10:  3ffeb82a 3fff23d0 3fff2a8c 40238d45  
3fffff20:  3fff2104 3fff2c44 3fff2c44 3ffef490  
3fffff30:  00000000 3fff2a8c 0000001c 3fff2c44  
3fffff40:  3ffeb81c 00000000 3fff2a8c 40238141  
3fffff50:  0700000a 000001ce 00000000 00000014  
3fffff60:  00000002 0000001a 40218b8b 3ffee370  
3fffff70:  3ffeb7f4 3fffdcc0 3ffea9c0 3ffea9c0  
3fffff80:  40218afe 3ffee370 00000000 3fff2c94  
3fffff90:  3fffdc80 00000000 3fff2a8c 4022cbcf  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
…CáD: FS Total 2949250, used 14056
`

Edzelf, 2016-11-15:
I think that in an asynchronous part of the code (the webserver) a  "yield" or "delay" is executed.  That is not allowed.  Does the radio works if you do not connect to the webserver?

Pavulon87, 2016-11-24:
Got the same problem. It is caused by lauching "list" command.
It happens when **radio.ini** is not ended with new line character.
In `readinifile()` function:
`line = inifile.readStringUntil ( '\n' ) ;`

Pavulon87, 2016-11-24:
Quick workaround:

`
void getpresets ( AsyncWebServerRequest *request )
{
  String              path ;                           // Full file spec as string
  File                inifile ;                        // File containing URL with mp3
  String              line ;                           // Input line from .ini file
  String              linelc ;                         // Same, but lowercase
  int                 inx ;                            // Position of search char in line
  int                 i ;                              // Loop control
  AsyncResponseStream *response ;                      // Response to client

  response = request->beginResponseStream ( "text/plain" ) ;
  path = String ( INIFILENAME ) ;                      // Form full path
  inifile = SPIFFS.open ( path, "r" ) ;                // Open the file

  if ( inifile )
  {
    int fileSize = inifile.size();
    int readBytes = 0;
    char c[1];
    
    while ( inifile.available() && readBytes < fileSize  )
    {
      line = "";
      while ( readBytes < fileSize && inifile.readBytes(c, 1) )
      {
        readBytes++;
        
        if ( c[0] == '\n' )
        {
          break;
        }
        
        line += c[0];
      }
      
      //line = inifile.readStringUntil ( '\n' ) ;        // Read next line
      linelc = line ;                                  // Copy for lowercase
      linelc.toLowerCase() ;                           // Set to lowercase
      if ( linelc.startsWith ( "preset_" ) )           // Found the key?
      {
        i = linelc.substring(7, 9).toInt() ;           // Get index 00..99
        inx = line.indexOf ( "#" ) ;                   // Get position of "#"
        if ( inx > 0 )                                 // Equal sign present?
        {
          line.remove ( 0, inx + 1 ) ;                 // Yes, remove non-comment part
        }
        else
        {
          inx = line.indexOf ( "=" ) ;                 // Get position of "="
          line.remove ( 0, inx + 1 ) ;                 // Yes, remove first part of line
        }
        line = chomp ( line ) ;                        // Remove garbage
        response->printf ( "%02d%s|", i,
                           line.c_str() ) ;            // 2 digits plus description
      }
    }
    inifile.close() ;                                  // Close the file
  }
  
  request->send ( response ) ;
}
`

jammoyano, 2017-02-12:
Hi guys,

Sorry for too long answer, but I've several "questions" to solve (including moved to another city)
I've just try the new web interface by @Sander0542 (https://github.com/Edzelf/Esp-radio/issues/52) and seems to be solved ... but It still in "surveillance mode"

Thanks.

closed 2017-02-12 by jammoyano