Edzelf/Esp-radio #41: Way to gain extra pins

fredericplante, 2016-11-15:
Has any one consider introducing a PCF8574 chip to handle all the control lines? The chip even has an interrupt pin so the handling of DREQ on the vs1053 would be fairly générique. I might give it a look this weekend

Have a good one you all :D

Edzelf, 2016-11-15:
I did some experimenting with a simple 74HC595.  With this shiftregister connected to the SPI-bus you can control 8 output lines including:
VS1053: Reset, DCS and CS
TFT:  Reset, D/C and CS
The rest of the outputs may be used  to drive some informative LEDs.

fredericplante, 2016-11-15:
The problem with 595 is that they use an cs line of it's own. while the pcf8574 only use 2, the extra one is to probe interruption, if you want it. otherwise you can drop the int line and check the I2C when you feel it need it. 

Many people don't like the I2C for many reason, the fact is that it's no good for heavy stream of data. But it's the best to handle small stuff, like doing an interrupt contrôler, keyboard, status led, and other simple data.

68erDirk, 2016-11-15:
Why don't you spent a bit more than 2 Euros and buy an arduino pro mini or nano? The ka-radio project uses an arduino nano just connected by the TX Wire, VCC and GND to connect to a display. It gives a lot of additional possibilities like easily using a remote control.

fredericplante, 2016-11-15:
Bravo 68erDirk, your project also rock :)

68erDirk, 2016-11-15:
it's not my project, it is karawin's project. I just built both since I wanted to find out the differences. I like both. Ka-radio has the advantage of downloading new versions via OTA and it has a nicer web interface. The new idea of an additional idea seems promising for add-ons. Your project is great since it is very minimalistic (display without additional chip), so I built it into a very small case of about 4x5x3 cm size! I prfer the Arduino IDE since I have some experience with it and it was not necessary to create a different ide. So I was able to create an addon (see my fork with pull request) to use the single button for three features (Fwd, Back an d mute). I would like to contribute more features to your project, so if you have some ideas, feel free to ask me for help.

fredericplante, 2016-11-15:
The next step, I guess, is to plug all that on the TVOut project

closed 2016-12-12 by fredericplante