Edzelf/Esp-radio #42: App

Sander0542, 2016-11-23:
Hello,

This is not an issue, but i'am making an app and I was wondering if I have to use the buttons or can use some urls like <ip>/index.html?<parameter>=<value>
Is this possible, because I dont have the radio right now.

Kinds Regards,
Sander Jochems

Edzelf, 2016-11-24:
Sander,
Yes, you can control the radio through the web-interface.  That's the way I use it: no buttons.  You may also control it by MTTQ commands or through the serial interface, although the last option is probably not suitable in combination with an app.

Regards,
Ed.

Sander0542, 2016-11-24:
Thanks,

Could you please give me an list of all the url parameters so i can make as much as possible in the app

Kind Regards,
Sander Jochems

Edzelf, 2016-11-25:
It is documented in the pdf-file.

Sander0542, 2016-11-26:
In the PDF is see ?parameter=value but i can get anymore out

I have this:
?uppreset=1
?downpreset=1
?upvolume=2
?downvolume=2

Sander0542, 2016-11-26:
Is there a way to add a station without entering a number or can I look what the latest number is?

Sander0542, 2016-12-01:
If you want to take a look on what I have now visit [this site](http://www.sanderjochems.nl/projects/2). Any ideas or tips are welcome.

Sander0542, 2016-12-06:
I have added a download link on the site, if you want to try it your welcome to do. if you find any bug could you please tell them.

Edzelf, 2016-12-07:
I used in the software the settings according to the datasheet:
- ST AMPLITUDE bits 15:12 Treble Control in 1.5 dB steps (-8..7, 0 = off)
- ST FREQLIMIT bits 11:8 Lower limit frequency in 1000 Hz steps (1..15)
- SB AMPLITUDE bits 7:4 Bass Enhancement in 1 dB steps (0..15, 0 = off)
- SB FREQLIMIT bits 3:0 Lower limit frequency in 10 Hz steps (2..15)

Although 4 bits are reserved for the 4 settings, not all 16 values are meaningful.  The select buttons in index.html are used to limit the settings to acceptable values (14 to 16 options per function, never 17).

GabeHC, 2017-01-20:
preset station not working which has Chinese characters in it's name. 
help please!

Edzelf, 2017-01-20:
Example please.

Verzonden vanaf Samsung-tablet.

Sander0542, 2017-01-20:
Maybe the characters are not supported by the app, if you send an example i can test it and fix this bug.

GabeHC, 2017-01-21:
I am using Lenovo YOGA tablet. my radio.ini is [Here](https://dl.dropboxusercontent.com/u/23181672/radio.ini)

Sander0542, 2017-01-21:
Hello, 

There was a bug in the app with preset numbers, if you update the app then I guess it is fixed.

Regards,
Sander

GabeHC, 2017-01-21:
did you update download link in your project?
try to update several times, still V 1.3.4

Sander0542, 2017-01-21:
I'll reupload the file

Sander0542, 2017-03-28:
The App is now uploaded to Google Play --> [](https://play.google.com/store/apps/details?id=com.thunkable.android.sander542jochems.ESP_Radio)

Its also updated on the about page.

Sander0542, 2017-04-06:
If anyone has good ideas for the app, just say it to me and I may can add it!