Edzelf/Esp-radio #43: local mp3 file

medamarko, 2016-11-26:
Is it possible to play local mp3 file that is located in spiffs?

fredericplante, 2016-11-26:
You plan to put 1 song in your SPIFFs? 

Remember at 128kb/s a mp3 is more or less1mB per minute. So you have about 3 minutes of space available in flash

medamarko, 2016-11-26:
How this can be done, by sending url of its own with path to mp3 on spiffs?
Yeah i just want to put couple short under 15 sec files for tests. Main reason is to adjust eq to get some nice sound as well :)

fredericplante, 2016-11-26:
Well then just use your SPIFFS like if it was an SD card, the 2 librarys pretty much work the same way. So what you would do to load it from SD, do it from SPIFFS

Edzelf, 2016-11-28:
I did have a version of ESP-Radio that played mp3-file from SPIFFS.  Th SPIFFS is not very reliable when reading big files: the data rate is sometimes slowed down at a certain position in the file.
But I will implement this feature in the next release.  URLs for local files will start with "localhost...".

medamarko, 2016-12-16:
that would be very helpful. thanks!

Edzelf, 2016-12-23:
This feature is implemented in the 23-12-2016 version.

closed 2016-12-23 by Edzelf