Edzelf/Esp-radio #44: Version without VS1053

probonopd, 2016-12-03:
Is is also possible to run this without the VS1053 as described [here](https://github.com/espressif/ESP8266_MP3_DECODER#running-without-the-i2s-dac)? This would shorten the long waiting time for the China module to arrive...

probonopd, 2016-12-03:
Similarly, I am wondering whether a "[Audio AUX WITH IR Remote Control MP3+WAV+WMA Decoder Board 2W Amplifier TF Card](http://www.ebay.com/itm/Audio-AUX-WITH-IR-Remote-Control-MP3-WAV-WMA-Decoder-Board-2W-Amplifier-TF-Card/131851346950)" could somehow be hooked up to the ESP. For 5 USD this not only includes a MP3 decoder, but also an infrared remote sender and receiver, and a class D amplifier.

Question, how would we get the radio streams in there. Emulate reading an audio file from SD card?

Edzelf, 2016-12-05:
I don't think it is worthwhile to use the PWM method.  The sound quality is minimal and stereo is not possible.
The GPD2826C chip is great, but it has no interface to communicate with the ESP8266.

probonopd, 2016-12-05:
So if we could emulate a SD card, the GPD2826C could "read" from it...

fredericplante, 2016-12-06:
That is actually the most retort idea I have read so far. Can the SPI library allow the esp8266 to work in SPI slave mode? Then the next question is, are you willing to make a FAT emulator that will patch between the Net and the GPD2856C module to make a stream flow so the GPD2856C believe he is dealing with a SDcard.

fredericplante, 2016-12-06:
I guess, if you manage to put the ESP8266 in SPI slave mode, It would be a breeze to simply wait for the play signal, then upload the MP3 crudely, like you do with the VS1053, straight to the GPD2856C module, using the SD port. I would drop functions like next and last. button. 

I would be curious to see if it's very complicated to pick up the recorded signal, using the record function on the GPD2856C module, to stream the input line of the device over the network

fredericplante, 2016-12-06:
http://electronics.stackexchange.com/questions/29010/sd-card-emulation

Edzelf, 2016-12-07:
It's a wild idea...  But I don't think I will pick this up.
Even wilder: There are SD-cards that can be filled over a WiFi connection, so theoretically you could send mp3 files OTA to the SD in the module....

milkpirate, 2017-05-16:
Since we are talking about alternatives to the VS1053: As the, by probonopd, mentioned project uses an 
 I2S DAC (ES9023)... I have laying around. Since I had trouble getting this setup to work I would like to try you code with the ES9023. Would that be suitable to your project?

fredericplante, 2017-05-16:
The EsP would require some  extra RAM. There is such an MP3 player on the esp github page, I think

fredericplante, 2017-05-16:
Does not seem to be there

Edzelf, 2017-05-16:
I'm working on the ESP32 version for this radio.  That will give you plenty of RAM.

fredericplante, 2017-05-16:
Oh I have not follow for a month, now, so it work on esp32 now? I have lot of reading to do, it seem.