Edzelf/Esp-radio #45: Some error message

fredericplante, 2016-12-08:
I finally got to the point where it was your project turn to try. But I got these error message, myust be nothing but...

>  Arduino: 1.6.5 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 921600, 4M (3M SPIFFS), ck, Disabled, None"
> 
> In file included from C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.h:25:0,
>                  from C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,
>                  from Esp_radio.ino:115:
> c:\documents and settings\so sory\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
> C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
>  #define max(a,b) ((a)>(b)?(a):(b))
>                   ^
> c:\documents and settings\so sory\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':
> C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
>  #define max(a,b) ((a)>(b)?(a):(b))
>                   ^
> Error compiling.
> 
>   This report would have more information with
>   "Show verbose output during compilation"
>   enabled in File > Preferences.
>

Edzelf, 2016-12-08:
Did you change something around line 112?  It should read something like:
// Define the version number:
#define VERSION "15-nov-2016"
// TFT.  Define USETFT if required.
#define USETFT
#include <ESP8266WiFi.h>

fredericplante, 2016-12-08:
`// 15-11-2016, ES: Support for .m3u playlists.
//
// Define the version number:
#define VERSION "15-nov-2016"                  //My line 112
// TFT.  Define USETFT if required.
#define USETFT
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncMqttClient.h>
#include <SPI.h>
#if defined ( USETFT )
#include <Adafruit_GFX.h>
#include <TFT_ILI9163C.h>
#endif`

fredericplante, 2016-12-08:
// 15-11-2016, ES: Support for .m3u playlists.
//
// Define the version number:
#define VERSION "15-nov-2016"           /<--- Line 112
// TFT.  Define USETFT if required.
#define USETFT
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncMqttClient.h>
#include <SPI.h>
#if defined ( USETFT )
#include <Adafruit_GFX.h>
#include <TFT_ILI9163C.h>
#endif

fredericplante, 2016-12-08:
If it can help
[image]

fredericplante, 2016-12-08:
This is what comes out with full débug message:

> Arduino: 1.6.5 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (3M SPIFFS), ck, Disabled, None"
> 
> Build options changed, rebuilding all
> 
> In file included from Esp_radio.ino:117:0:
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:310:18: warning: unused parameter 'request' [-Wunused-parameter]
>      virtual bool canHandle(AsyncWebServerRequest *request){ return false; }
>                   ^
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:311:18: warning: unused parameter 'request' [-Wunused-parameter]
>      virtual void handleRequest(AsyncWebServerRequest *request){}
>                   ^
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:312:18: warning: unused parameter 'request' [-Wunused-parameter]
>      virtual void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final){}
>                   ^
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:312:18: warning: unused parameter 'filename' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:312:18: warning: unused parameter 'index' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:312:18: warning: unused parameter 'data' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:312:18: warning: unused parameter 'len' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:312:18: warning: unused parameter 'final' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:313:18: warning: unused parameter 'request' [-Wunused-parameter]
>      virtual void handleBody(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total){}
>                   ^
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:313:18: warning: unused parameter 'data' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:313:18: warning: unused parameter 'len' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:313:18: warning: unused parameter 'index' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:313:18: warning: unused parameter 'total' [-Wunused-parameter]
> In file included from C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:403:0,
>                  from Esp_radio.ino:117:
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/WebResponseImpl.h:41:20: warning: unused parameter 'buf' [-Wunused-parameter]
>      virtual size_t _fillBuffer(uint8_t *buf, size_t maxLen){ return 0; }
>                     ^
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/WebResponseImpl.h:41:20: warning: unused parameter 'maxLen' [-Wunused-parameter]
> In file included from C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:405:0,
>                  from Esp_radio.ino:117:
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:58:18: warning: unused parameter 'len' [-Wunused-parameter]
>      virtual void ack(size_t len, uint32_t time){}
>                   ^
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:58:18: warning: unused parameter 'time' [-Wunused-parameter]
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:59:20: warning: unused parameter 'client' [-Wunused-parameter]
>      virtual size_t send(AsyncClient *client){ return 0; }
>                     ^
> In file included from C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.h:25:0,
>                  from C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,
>                  from Esp_radio.ino:115:
> c:\documents and settings\so sory\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
> C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
>  #define max(a,b) ((a)>(b)?(a):(b))
[... cut ...]
>  static bool ON_STA_FILTER(AsyncWebServerRequest *request) {
>              ^
> C:\Documents and Settings\So sory\My Documents\Arduino_ESP\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:267:13: warning: 'bool ON_AP_FILTER(AsyncWebServerRequest*)' defined but not used [-Wunused-function]
>  static bool ON_AP_FILTER(AsyncWebServerRequest *request) {
>              ^
> Error compiling.
> 
>   This report would have more information with
>   "Show verbose output during compilation"
>   enabled in File > Preferences.
>

fredericplante, 2016-12-08:
I don't think that the problem is in your code, the problem seem to come from the AsynMTTQclient library, when I try ton compile the FullyFeatured example, it return the same exact message:

> Arduino: 1.6.5 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (3M SPIFFS), ck, Disabled, None"
> 
> In file included from C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.h:25:0,
>                  from C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,
>                  from FullyFeatured.ino:1:
> c:\documents and settings\so sory\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
> C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
>  #define max(a,b) ((a)>(b)?(a):(b))
>                   ^
> c:\documents and settings\so sory\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':
> C:\Documents and Settings\So sory\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/Arduino.h:253:18: error: expected unqualified-id before '(' token
>  #define max(a,b) ((a)>(b)?(a):(b))
>                   ^
> Error compiling.
> 
>   This report would have more information with
>   "Show verbose output during compilation"
>   enabled in File > Preferences.

Any suggestions?

Edzelf, 2016-12-08:
Reïnstall IDE and libraries?

Verzonden vanaf Samsung-tablet.

medamarko, 2016-12-09:
I would try to take ARDUINO directly from git and copy over installed one, there were some fixes in recent weeks.
https://github.com/esp8266/Arduino

fredericplante, 2016-12-09:
> Sketch uses 354,961 bytes (33%) of program storage space. Maximum is 1,044,464 bytes.
> Global variables use 44,072 bytes (53%) of dynamic memory, leaving 37,848 bytes for local variables. Maximum is 81,920 bytes.

Ah ah! This only show that I've negleted my arduino install folder for to long. ;)

Thank you for you patience. I getting on the few implementation that I was thinking about.

fredericplante, 2016-12-09:
By the way Would you freak alot if I put a toggle on the debug output to redirected to an telnet kind of output? Cause I would really appreciate to regain GPIO1 and GPIO3 to setup an I2C interface

closed 2016-12-09 by fredericplante

Edzelf, 2016-12-09:
No problem, go ahead.

Verzonden vanaf Samsung-tablet.