Edzelf/Esp-radio #46: MetaName

Sander0542, 2016-12-10:
Hello,

Is there a way to get the song title in the website?

fredericplante, 2016-12-11:
There must be some standard in streaming MP3, just like in normal .mp3, there must be idx tag, or some thing.

[ID tag](https://www.google.ca/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwj19_e_luvQAhUB62MKHYr2AEsQFggdMAA&url=https%3A%2F%2Ffr.wikipedia.org%2Fwiki%2FID3&usg=AFQjCNGIy2KYXBVWVL5WSmX13Fum-KemjQ&sig2=5N63BOOQhxtJNbJ1SJIlqg&bvm=bv.141320020,d.cGc)

Sander0542, 2016-12-11:
Yes, but if you have debug on then you can see the StreamTitle but in the website i cant find any thing about it

fredericplante, 2016-12-11:
Oh I get what you mean, yeap, many thing miss on the web page, I say, it also lack a play, pause, stop buttons. and the web interface to play a standalone mp3 file. I let you work on this while I'm on the I2C interface to gain extra pins

Edzelf, 2016-12-12:
I will include some extra song information (web interface) in the next release.  In the current version this is only visible on the TFT.

fredericplante, 2016-12-12:
A webpage containing a play list that point to stand alone  mp3 would be nice also. ;)

Edzelf, 2016-12-23:
The status button displays some info now.  Will be better in the future.

closed 2016-12-26 by Sander0542