Edzelf/Esp-radio #51: Resume player after stop command

ohalushchak, 2016-12-25:
Tried to add a new command "resume". It should start to play music after it was stoped. For this command I do the next logic: 

hostreq = true ;

But it looks like something else is missing.

Edzelf, 2016-12-25:
Looks good.  I will test it later.

Verzonden vanaf Samsung-tablet.

ohalushchak, 2016-12-25:
Here is some additional information:
Request: /?stop=0
log: 
D: Command: stop with parameter 0
D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec

Request: /?resume=0
log:
Command: resume with parameter 0
D: Connect to new host stream.mjoy.ua
D: Connect to stream.mjoy.ua on port 80, extension /
D: Connected to server
D: Server: Apache/2.4.7 (Ubuntu)
D: Last-Modified: Fri, 16 Oct 2015 10:26:07 GMT
D: Accept-Ranges: bytes
D: Content-Length: 11510
D: Connection: close
D: Content-Type: text/html
D: Switch to DATA, bitrate is 0
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
D: 
D: 
D: Song stopped correctly after 0 msec
D: New preset/file requested
D: Connect to new host ua.uar.net:8000/galychyna
D: Slash in station
D: Connect to ua.uar.net on port 8000, extension /galychyna
D: Connected to server
D: Server: Icecast 2.4.1
D: Content-Type: audio/mpeg
D: Cache-Control: no-cache
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Pragma: no-cache
D: Access-Control-Allow-Origin: *
D: icy-br:256
D: icy-description:This is only a trial
D: icy-genre:galychyna
D: icy-name:Galychyna FM
D: icy-pub:0
D: icy-url:http://www.galychyna.fm
D: icy-metaint:16000
D: Switch to DATA, bitrate is 256
D: Metadata block 16 bytes
D: Streamtitle found, 15 bytes
D: StreamTitle='';
D: Recovered from dataloss

Edzelf, 2016-12-26:
Yes, it's clear.  The variable "host" is shortened in the function "connecttohost", making it unusable for a second call to this function.  I will change this and add the "resume" feature to the software within a few days.

Edzelf, 2016-12-28:
Resume works now.  It is not yet implemented in the web interface.

closed 2016-12-28 by Edzelf