Edzelf/Esp-radio #52: Website

Sander0542, 2016-12-26:
Hello,

I have created an website interface that looks better, I'll include a download once I'm done with testing.

Regards,
Sander Jochems

Sander0542, 2016-12-26:
I couldn't found any errors. here is the zip with all the files in the data folder
[data.zip](https://github.com/Edzelf/Esp-radio/files/673111/data.zip)

grzegorzshub, 2016-12-27:
Hello
Looks much better indeed. Any chance to get an offline version and ESP fs friendly please?
Best wishes
Grzegorz

Sander0542, 2016-12-28:
Hello,

Could you tell me what you mean with an offline version?
I don't know what a ESP fs friendly is.

Regards,
Sander

grzegorzshub, 2016-12-28:
Hi
I use isolated network with local icecast server without internet access. It looks like your nice pages need access to internet. I was thinking if I could download external files down to ESP fs.
Best
Grzegorz

Sander0542, 2016-12-28:
Yes, i use some internet css for the style. I will include the css in the data map

Sander0542, 2016-12-28:
I'll add a new donwload link in a few minutes

grzegorzshub, 2016-12-28:
Amazing! Thank you, much apriciate.
Best wishes
Grzegorz

Sander0542, 2016-12-28:
I'm done, it should work now
[data.zip](https://github.com/Edzelf/Esp-radio/files/675439/data.zip)

Edzelf, 2016-12-28:
I like looks!  But 1.5 MB is quite a lot of space...

Sander0542, 2016-12-28:
Yes, that is because the css is in the data map, if you take remove the css map and change te url its a lot smaller

fredericplante, 2016-12-28:
Personally, I was planing to make a SD http serveur, and let the radio module connect on it for local .mp3 playing. Cause if I remember well, the VS10x3 don't like to be on the same bus the the SDcard reader, some conflict on MISO line, unless you uuse a 3states buffer to Isolate the sdCard.

Sander0542, 2016-12-28:
I have two version now:

Internet CSS --> [data.zip](https://github.com/Edzelf/Esp-radio/files/676121/data.zip)
Included CSS -->[data.zip](https://github.com/Edzelf/Esp-radio/files/676120/data.zip)

I would use the Internet CSS, becuase then you have more space left in your ESP8266.

Edzelf, 2016-12-29:
I like the "Included" version for safety and independency.  I will try to make a smaller version with the same looks.  My aim is 95 percent of the looks for 1 percent of the size...

Sander0542, 2016-12-29:
Yes, the best thing you should do is to remove all the parts from the css which are not used.

Edzelf, 2016-12-29:
New interface now in data-directory (14 KB).

Sander0542, 2016-12-30:
Nice work,

There are a few things I would change to make it even nicer.

- White background of the textboxes
- No URL in the "ESP Radio" title in the topbar
- 3 Links on every page, not 2

If you want, i can change it and send it for the next update

Sander0542, 2016-12-30:
I'm done, [data.zip](https://github.com/Edzelf/Esp-radio/files/678806/data.zip) (17,1 kB)

thermo2016, 2016-12-30:
I think the white background & links etc can be fixed by editing the ESP-radio.ino file.
Go to function String getContentType ( String filename )
and add this line:

`else if ( filename.endsWith ( ".css"  ) ) return "text/css" ;`

Without this, the esp internal web sever returns ccs code as text/plain whereas it needs to be text/ccs.
I hope this helps.

Edzelf, 2016-12-31:
I added Sanders website to the data directory.
I added the ContentType "text/css" to the sketch.

Edzelf, 2017-01-02:
The webinterface is now in PROGMEM.  More reliable than in SPIFFS.

Edzelf, 2017-01-03:
I found out that the ESP8266 will not handle multiple connections to the async webserver.  Most browsers will open multiple connections to get index.html, radio.css and favicon.ico simultaneously.  In Mozilla I changed the "network.http.max-persistent-connections-per-server"to 1 (was 6).

Pavulon87, 2017-01-06:
Edzelf, maybe replace `<link rel="stylesheet" type="text/css" href="radio.css">` with
`<script type="text/javascript">
var stylesheet = document.createElement('link');
stylesheet.href = 'radio.css';
stylesheet.rel = 'stylesheet';
stylesheet.type = 'text/css';
document.getElementsByTagName('head')[0].appendChild(stylesheet); 
</script>`
and add `<noscript><link rel="stylesheet" href="radio.css"></noscript>` after `</html>`
It should work.

Edzelf, 2017-01-06:
I will try that.  Can you explain the difference?

Pavulon87, 2017-01-07:
radio.css should start loading after page is loaded, so multiple connections will not be needed.
Tested in google chrome and firefox

Pavulon87, 2017-01-07:
I'm also trying additional headers.
changed(every file):
`request->send_P( 200, ct, index_html );`
to:
`AsyncWebServerResponse *response = request->beginResponse_P( 200, ct, index_html ) ; `
and added:
`response->addHeader("Server","Esp-radio Web Server");`
`response->addHeader("Cache-Control","max-age=3600");`
`response->addHeader("Last-Modified","Sat, 07 Jan 2017 21:00:00 GMT");`
`request->send(response);`

Edzelf, 2017-01-16:
Pavulon87:  I updated the project according to you suggestions.

GabeHC, 2017-01-20:
What tool should I use to convert those resource files into C array?
I had custom icon need to put into PROGMEM. Don't know how to do it.

Edzelf, 2017-01-20:
I used "od" on a Linux machine, but Googling on "dump file in ascii" or "hexdump" will give you all the answers.