Edzelf/Esp-radio #55: ESP radio does not play after update

fh19, 2017-01-14:
Hello,

great project; it worked fine with the version from Nov 15, now I updated to the current version (Jan 2) and it does no longer play any radio station. It always tries until it reboots, but never succeeds. Going back to the old version works fine (w/ identical ini-file). What could that be?

Thanks for any help.

Debug-output is:

> D: FS Total 52961, used 14809
> D: /index.html                      -    6771
> D: /config.html                     -    3700
> D: /favicon.ico                     -     766
> D: /radio.ini                       -    2105
> D: Added SSID 00 = EasyBox-Hk to acceptable networks
> D: * Scan Networks *
> D: Number of available networks: 3
> D:  1 - EasyBox-Hk                Signal: -76 dBm Encryption Auto  Acceptable
> D: --------------------------------------
> D: Command: mqttbroker with parameter broker.hivemq.com
> D: Command: mqttport with parameter 1883
> D: Command: mqttuser with parameter none
> D: Command: mqttpasswd with parameter none
> D: Command: mqtttopic with parameter espradio
> D: Command: mqttpubtopic with parameter espradioIP
> D: Command: wifi_00 with parameter EasyBox-Hk/XXXXXXXX
> D: Command: volume with parameter 80
> D: Command: toneha with parameter 0
> D: Command: tonehf with parameter 0
> D: Command: tonela with parameter 0
> D: Command: tonelf with parameter 0
> D: Command: preset with parameter 0
> D: Preset set to 0
> D: Command: preset_00 with parameter mp3-live.swr3.de:80/swr3_m.m3u
> D: Command: preset_01 with parameter mp3-live.swr.de:80/swr1bw_m.m3u
> D: Command: preset_02 with parameter streams.br.de:80/bayern3_2.m3u
> D: Command: preset_03 with parameter metafiles.gl-systemhaus.de:80/hr/hr3_2.m3u
> D: Command: preset_04 with parameter energyradio.de:80/stuttgart
> D: Command: preset_05 with parameter 109.206.96.34:8100
> D: Command: preset_06 with parameter airspectrum.cdnstream1.com:8114/1648_128
> D: Command: preset_07 with parameter us2.internet-radio.com:8050
> D: Command: preset_08 with parameter airspectrum.cdnstream1.com:8000/1261_192
> D: Command: preset_09 with parameter airspectrum.cdnstream1.com:8008/1604_128
> D: Command: preset_10 with parameter us1.internet-radio.com:8105
> D: Command: preset_11 with parameter icecast.omroep.nl:80/radio1-bb-mp3
> D: Command: preset_12 with parameter 205.164.62.15:10032
> D: Command: preset_13 with parameter skonto.ls.lv:8002/mp3
> D: Command: preset_14 with parameter 94.23.66.155:8106
> D: Starting ESP Version 02-jan-2017...  Free memory 14464
> D: Sketch size 345896, free size 90112
> D: Reset VS1053...
> D: End reset VS1053...
> D: Slow SPI,Testing VS1053 read/write registers...
> D: Fast SPI, Testing VS1053 read/write registers again...
> D: endFillByte is 0
> D: Selected network: EasyBox-Hk               
> D: Try WiFi EasyBox-Hk
> D: IP = 192.168.0.95
> D: Start server for commands
> D: Connecting to MQTT broker.hivemq.com, port 1883, user none, password none...
> D: MQTT Connected to the broker broker.hivemq.com, session is not present
> D: Subscribing to espradio at QoS 2, packetId = 1 
> D: Publishing IP 192.168.0.95 to topic espradioIP
> D: MQTT Disconnected from the broker, reason 0, reconnecting...
> D: STOP requested
> D: Song stopped correctly after 0 msec
> D: New preset/file requested
> D: Connect to new host mp3-live.swr3.de:80/swr3_m.m3u
> D: Playlist request, entry 1
> D: Connect to mp3-live.swr3.de on port 80, extension /swr3_m.m3u
> D: Connected to server
> D: Server: Apache
> D: Content-Disposition: attachment; filename="swr3_mp3_m.m3u"
> D: X-Playlist-Cache-Modified: Fri, 24 Jun 2016 10:15:18 GMT
> D: Content-Length: 143
> D: Content-Type: audio/x-mpegurl
> D: Cache-Control: max-age=86400
> D: Connection: close
> D: Switch to DATA, bitrate is 0
> D: First chunk:
> D: 23 45 58 54 4D 33 55 0A
> D: 23 45 58 54 49 4E 46 3A
> D: 2D 31 2C 53 57 52 33 2D
> D: 4C 69 76 65 73 74 72 65
> D: No data input
> D: Trying other station/file...
> D: STOP requested
> D: Song stopped correctly after 0 msec
> D: New preset/file requested
> D: Connect to new host mp3-live.swr3.de:80/swr3_m.m3u
> D: Playlist request, entry 2
> D: Connect to mp3-live.swr3.de on port 80, extension /swr3_m.m3u
> D: Connected to server
> D: Server: Apache
> D: Content-Disposition: attachment; filename="swr3_mp3_m.m3u"
> D: X-Playlist-Cache-Modified: Fri, 24 Jun 2016 10:15:18 GMT
> D: Content-Length: 143
> D: Content-Type: audio/x-mpegurl
> D: Cache-Control: max-age=86400
> D: Connection: close
> D: Switch to DATA, bitrate is 0
> D: No data input
> D: Trying other station/file...
> D: STOP requested
> D: Song stopped correctly after 0 msec
> D: New preset/file requested
> D: Connect to new host mp3-live.swr3.de:80/swr3_m.m3u
> D: Playlist request, entry 3
> D: Connect to mp3-live.swr3.de on port 80, extension /swr3_m.m3u
> D: Connected to server
> D: Server: Apache
> D: Content-Disposition: attachment; filename="swr3_mp3_m.m3u"
> D: X-Playlist-Cache-Modified: Fri, 24 Jun 2016 10:15:18 GMT
> D: Content-Length: 143
> D: Content-Type: audio/x-mpegurl
> D: Cache-Control: max-age=86400
> D: Connection: close
> D: Switch to DATA, bitrate is 0
> D: No data input
> D: Trying other station/file...
> D: STOP requested
> D: Song stopped correctly after 0 msec
> ...

Edzelf, 2017-01-14:
Looks like the playlist feature is broken.  Could you try the .ini file with these presets:
# Presets
preset = 6						# Start with preset 6
preset_00 = 109.206.96.34:8100				#  0 - NAXI LOVE RADIO, Belgrade, Serbia
preset_01 = airspectrum.cdnstream1.com:8114/1648_128	#  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050			#  2 - CLASSIC ROCK MIA WWW.SHERADIO.COM
preset_03 = airspectrum.cdnstream1.com:8000/1261_192	#  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128	#  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105			#  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3		#  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032				#  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3			#  8 - Skonto 128k
preset_09 = 94.23.66.155:8106				#  9 - *ILR CHILL and GROOVE

I will try yours and see what the problem is.

fh19, 2017-01-15:
Hello,

thanks for the quick reply. Your presets work; if I re-enter even some 
of my presets, it stucks again. could it be a problem with the m3u-files 
requested by e.g. SWR3?

Best regards,
Falk

Edzelf, 2017-01-16:
I have corrected the bug with m3u-files in the 17-jan-2017 version.

fh19, 2017-01-16:
Hi,

thanks again for the quick fix. I've tried it and m3u radio stations work fine. But I got a crash when I call in a web-browser the config page. It seems that it is caused in the function void handleFSf (...); when I modified it like below it worked (moving the addHeader calls inside the last else block), even though I'm not sure about the reason:

```

//******************************************************************************************
//                                H A N D L E F S F                                        *
//******************************************************************************************
// Handling of requesting files from the SPIFFS/PROGMEM. Example: /favicon.ico             *
//******************************************************************************************
void handleFSf ( AsyncWebServerRequest* request, String filename )
{
  String          ct ;                           // Content type
  AsyncWebServerResponse *response ;                    // For extra headers

  dbgprint ( "FileRequest received %s", filename.c_str() ) ;
  ct = getContentType ( filename ) ;                    // Get content type
  if ( ct == "" )                                       // Empty is illegal
  {
    request->send ( 404, "text/plain", "File not found" ) ;
  }
  else
  {
    if ( filename.indexOf ( "index.html" ) >= 0 )       // Index page is in PROGMEM
    {
      response = request->beginResponse_P ( 200, ct, index_html ) ;
    }
    else if ( filename.indexOf ( "radio.css" ) >= 0 )   // CSS file is in PROGMEM
    {
      response = request->beginResponse_P ( 200, ct, radio_css ) ;
    }
    else if ( filename.indexOf ( "config.html" ) >= 0 ) // Confige page is in PROGMEM
    {
      response = request->beginResponse_P ( 200, ct, config_html ) ;
    }
    else if ( filename.indexOf ( "about.html" ) >= 0 )  // About page is in PROGMEM
    {
      response = request->beginResponse_P ( 200, ct, about_html ) ;
    }
    else if ( filename.indexOf ( "favicon.ico" ) >= 0 ) // Favicon icon is in PROGMEM
    {
      response = request->beginResponse_P ( 200, ct, favicon_ico, sizeof ( favicon_ico ) ) ;
    }
    else
    {
      response = request->beginResponse ( SPIFFS, filename, ct ) ;
      //      request->send ( response ) ;
      // Add extra headers
      response->addHeader ( "Server", NAME ) ;
      response->addHeader ( "Cache-Control", "max-age=3600" ) ;
      response->addHeader ( "Last-Modified", VERSION ) ;
    }
    request->send ( response ) ;
    dbgprint ( "Response sent" ) ;
  }
}

```

Edzelf, 2017-01-17:
Bug fixed in 17-jan-2017 version.

closed 2017-01-19 by fh19