Edzelf/Esp-radio #56: geeetech vs1053 shield & wemos D1

thermo2016, 2017-01-15:
Hello, I was hoping to use one of these shields
http://www.geeetech.com/wiki/index.php/Arduino_MP3_shield_board_with_TF_card
fitted to a wemos D1. I've changed the digitial io lines to:
`#define VS1053_CS     D6`
`#define VS1053_DCS    D7`
`#define VS1053_DREQ   D2`

These values are taken from the (arduino) sparkfun libraries.
I've also undefined digital switches & tft to prevent any contention.
Unfortunately it doesn't work, I get these errors when starting:
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: VS1053 error retry SB:012C R1:0000 R2:0000
D: VS1053 error retry SB:0258 R1:0000 R2:0000
D: VS1053 error retry SB:0384 R1:0000 R2:0000
D: VS1053 error retry SB:04B0 R1:0000 R2:0000
D: VS1053 error retry SB:05DC R1:0000 R2:0000
D: VS1053 error retry SB:0708 R1:0000 R2:0000
D: VS1053 error retry SB:0834 R1:0000 R2:0000
D: VS1053 error retry SB:0960 R1:0000 R2:0000
D: VS1053 error retry SB:0A8C R1:0000 R2:0000
D: VS1053 error retry SB:0BB8 R1:0000 R2:0000
D: VS1053 error retry SB:0CE4 R1:0000 R2:0000
D: VS1053 error retry SB:0E10 R1:0000 R2:0000
D: VS1053 error retry SB:0F3C R1:0000 R2:0000
D: VS1053 error retry SB:1068 R1:0000 R2:0000
D: VS1053 error retry SB:1194 R1:0000 R2:0000
D: VS1053 error retry SB:12C0 R1:0000 R2:0000
D: VS1053 error retry SB:13EC R1:0000 R2:0000
D: VS1053 error retry SB:1518 R1:0000 R2:0000
D: VS1053 error retry SB:1644 R1:0000 R2:0000
D: VS1053 error retry SB:1770 R1:0000 R2:0000
D: VS1053 not properly installed!
D: endFillByte is 0

It looks like the vs1053 isn't being detected for some reason. 
Any ideas?
Thanks

GabeHC, 2017-01-16:
I have it figure out that WEMOS D1 map pin2 to ESP8266 GPIO 16, and it is the only pin can't be use for interrupt, So I had it remap like following:
#define VS1053_CS     16
#define VS1053_DCS    4
#define VS1053_DREQ   5

[[image]
](url)

and hardwire like this [image]

thermo2016, 2017-01-16:
Thank you for the very helpful info.  It's now working with the pins you suggested. Thank you.
(Although the issue is resolved the info is useful enough to keep somewhere)

GabeHC, 2017-01-20:
Dose any one gets SD files playing working?
Try to add SdFat.h SdFatUtil.h got an error says  redefinition of 'class fs::File'

thermo2016, 2017-01-21:
Are you using SDFat.h from the sparkfun libraries? If so, these do not compile on the esp8266 - I tried!
I have better luck when using an example from File->Examples->SD(esp8266).
Using the cardinfo example I was able to list the contents of the SD card - use chipSelect = 2;
If you have the sparkfun libraries, you may have to move/delete them.

GabeHC, 2017-01-21:
i had modify sparkfun libraries to work on ESP8266. and it works just fine.
the same hardware play MP3 files from SD card.
I thought it will be simple to just adapt it SdFat to ESP-RADIO
but there are conflicts on fs.h 
don't know differences between SPIFFS and SD file system. 
Do you?

acocalypso, 2017-03-16:
I have an geeetech vs1053 as well and using a nodemcu any idea how to get it working?

SurenBono, 2019-01-18:
> Hello, I was hoping to use one of these shields
> http://www.geeetech.com/wiki/index.php/Arduino_MP3_shield_board_with_TF_card
> fitted to a wemos D1. I've changed the digitial io lines to:
> `#define [...]

> `#define VS1053_CS D1`   // break pin xcs, hard solder vspin6 to D1 only after confirming 1ST  by fritzzing
> `#define VS1053_DCS D0` // break pin xds, hard solder vspin7 to D0 only after confirming 1ST  by fritzzing
> `#define VS1053_DREQ D2`  // remains

upload sketch with 180mhz & iW variant / v2HigherBandwidth option