Edzelf/Esp-radio #57: M3U not working.

GabeHC, 2017-01-17:
add these lines to preset station, but can't get it to work. 

preset_17 = pianosolo.streamguys.net/live.m3u               # 17 - Whisperings:Solo Piano Radio
preset_19 = listen.radionomy.com/taiwan-lounge.m3u    # 19 - Taiwan-Lounge Radio

Can you help? Thanks!

Edzelf, 2017-01-17:
Fixed.

GabeHC, 2017-01-17:
Thanks, Working good now!

is playing Youtube Radio possible?

Edzelf, 2017-01-17:
No, sorry.

arlaor, 2017-01-18:
Hi @Edzelf 
How can i play mp3 on a server?

Edzelf, 2017-01-19:
.mp3 should work.  Enter the following example in the in the play-field of the webinterface:
www.stephaniequinn.com/Music/Rondeau.mp3
or add it to your presets.
For (small) .mp3 file in SPIFFS use something like:
localhost/friendly.mp3

arlaor, 2017-01-19:
Hi @Edzelf 
Then what I would have to do is generate a local radio with my audio files?
Other Question. 
Is it possible to save the volume and equalization parameters for each radio station? Or does it save the general volume of last entry to the ESP-12E module?

Edzelf, 2017-01-19:
If you have your mp3-files on a NAS, you should be able to play them.
Parameters per station is a nice feature for the wishlist...

GabeHC, 2017-01-22:
My M3U file as following:
#EXTM3U
#EXTINF:281,JUDY10
http://10.0.0.5/music/2008/10.mp3
#EXTINF:282,JUDY02
http://10.0.0.5/music/2008/02.mp3
#EXTINF:236,JUDY03
http://10.0.0.5/music/2008/03.mp3
#EXTINF:257,JUDY04
http://10.0.0.5/music/2008/04.mp3
#EXTINF:254,JUDY05
http://10.0.0.5/music/2008/05.mp3
#EXTINF:258,JUDY06
http://10.0.0.5/music/2008/06.mp3
#EXTINF:268,JUDY07
http://10.0.0.5/music/2008/07.mp3
#EXTINF:247,JUDY08
http://10.0.0.5/music/2008/08.mp3
#EXTINF:326,JUDY09
http://10.0.0.5/music/2008/09.mp3
#EXTINF:219,JUDY01
http://10.0.0.5/music/2008/01.mp3

it keeps looping first track, is it something wrong it did of the m3u file or sketches bug?
and how to make it m3u8 compatible?

Edzelf, 2017-01-23:
Fixed.

arlaor, 2017-01-23:
Hi @Edzelf 

As I can set the VS1053 settings so that when I start, show them in the control page as I have set them in the radio.ini file, for example: treble gain + 4.5dB, Bass Freq: 50Hz.
I have tried to put them this way literally, but it does not show them as I have established them and I have also tried
```javascript
# VS1053 settings
volume = 84
toneha = 3
tonehf = 6
tonela = 5
tonelf = 6
```
But neither shows them and I do not know if they have already taken them internally.

another question. With what parameters and files do you recommend updating the firmware of the ESP-12E, in order to achieve the best compatibility between the module and the esp-radio project?

Edzelf, 2017-01-24:
You tone-settings should work.  But it is a one-way command.  The result is not visible in the webinterface.  Maybe I will include this feature in the next release.

As far as  I know, the term "firmware" is used in the ESP826 for the software in flash memory.  So in this case it's the software of Esp-radio.  Other types of "firmware" are NodeMCO or the "AT"-interface.
There is no way to update the  ROM-software.

arlaor, 2017-01-24:
Hi @Edzelf 
You can also include that automatically when the station is changed, by the NEXT or PREV button or when the module is turned on, automatically shows the station that is playing from the PRESET section and in the STATUS section, showing the streamtitle

GabeHC, 2017-01-26:
Sluggish WEB UI response when playing mp3, m3u.
Randomly skips songs playing.

arlaor, 2017-01-26:
sound error to station play http://live3.rcnmundo.com/lamega.mp3

Edzelf, 2017-01-26:
The station "live3.rcnmundo.com/lamega.mp3" gives the wrong stream info:
icy-metaint: 16000
In fact there is no metadata after 16000 bytes.
Look at the debug-output for more information.

arlaor, 2017-01-27:
Hi @Edzelf 
I was trying to get the debug-output for more information, but I did not know how to do it, the [information](https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md) I found about debug-output, talk about a debug port that my arduino IDE does not have and does not show how I can enable it to get it report.
The same problem I have obtained in other stations:
prem2.radiotunes.com/latinpophits_hi?6637214e119025aea9c9de18
19523.live.streamtheworld.com/LOS40.mp3
19373.live.streamtheworld.com/CADENADIAL.mp3
oidos.streamlicensing.com/thedumphigh?cb=472457.mp3

Which makes me believe that due to problems of speed of internal transmission of the ESP-12E or memory to buffer.
I do not know if the signatureware that has my module leaves very little work space for your sketch and must change it.

The arduino IDE also does not show exceptions that are generating the problem.
When the radio station generates the noises mentioned, the red led of the ESP-12E module flashes, indicating that there is a small cut in the transmission
What else can I check?

Edzelf, 2017-01-28:
prem2.radiotunes.com/latinpophits_hi?6637214e119025aea9c9de18 works well here.
19523.live.streamtheworld.com/LOS40.mp3 works well here.
19373.live.streamtheworld.com/CADENADIAL.mp3 works well here.
oidos.streamlicensing.com/thedumphigh?cb=472457.mp3 works well here.
live3.rcnmundo.com/lamega.mp3 still does not work due wrong "icy-metaint" supplied by the station.

For debugging: just start the "Serial monitor" of the IDE.  Thee first chunk of the mp3-data will be displayed and it should start with "FF FB" for these stations.

The ESP-8266 has enough memory and CPU power for this project, but it lacks buffers for multiple TCP-connections.  So when all inputbuffers are filled with mp3 data, you may have difficulties reaching the web interface.

The led on the ESP-module should be on (probably flashing very fast) all the time when playing.

arlaor, 2017-01-28:
`D: First chunk:
D: FF F3 82 64 00 0E E9 67
D: 75 3F 3C C2 6A 0F C8 FE
D: CA 44 4B C4 68 A1 5D E1
D: D5 12 D8 53 92 5B 1F A1
`

arlaor, 2017-01-29:
Hi @Edzelf 
I found the following when initially playing radio stations that have a bitrate of 64kbps and smaller bitrates.
Free memory is reduced to a value of 4000 and smaller, but when I initially touch stations with bitrates higher than 128kbps, free memory have a value of 8500 and higher.

When the free memory falls to a value of 4000, the stations begin to generate strange noises and to break the transmission. After lowering the free memory to 4000, it remains in this value and all the stations that are heard after this no matter the bitrate (128kbps), all produce strange noises and the free memory is not recovered until the reset button is pressed .

I do not believe that the error is of the radio transmitters but of the way the arduino code establishes the parameters to reproduce small bitrates and of not allow the recovery of the free memory or of releasing it after the change of a station of radio to another one.

Edzelf, 2017-01-29:
Can you give me the URL of the station that shows the problem?

arlaor, 2017-01-30:
live3.rcnmundo.com/ntn24.mp3
live3.rcnmundo.com/lamega.mp3
live4.rcnmundo.com/lafmpoplatino.mp3
live1.rcnmundo.com/rumbaybembe.mp3
...and others
These stations can listen to them normally without any noise or cuts from foobar2000 player or from google chrome, but unfortunately not in your project.
I would like to be able to solve these small problems in your project.

Another concern is about OTA. Request password to open the arduino IDE and to upload a file via the network port. what is the password?

Edzelf, 2017-01-30:
I will try.
No need for OTA password, just select the right port and start upload.  It 's not very reliable, sometimes it works after 3 times.

Edzelf, 2017-01-30:
All these station have an error in the metadata.  The first metadata block is always at 16014 bytes.  It should be at 16000.  After the first block it is sometimes at 16000 but mostly not.
I don't know a solution for this.

karawin, 2017-01-30:
Edzel, sorry but these stations are in chunked html. They play well in karadio.
See the chunked html definition at https://fr.wikipedia.org/wiki/Chunked_transfer_encoding
No doubt that you can improve your software.

karawin, 2017-01-30:
https://en.wikipedia.org/wiki/Chunked_transfer_encoding

Edzelf, 2017-01-30:
Clear.  It's new for me.  I will try to implement  this.

Edzelf, 2017-01-30:
Fixed.  New version supports "chunked transfer encoding" now.

arlaor, 2017-01-30:
Hi @Edzelf  @karawin 
Now it works normally. Excellent work

arlaor, 2017-01-30:
Hi @Edzelf 
The handling of the controls in the web page like volume, status, test, treble, Bass, when I am playing the radio stations mentioned in previous messages (64kbps and lower bitrates), produce important delays in the web page.

Edzelf, 2017-01-31:
See my answer on "Random error while updating web page".
There is a hardware solution: use a separate ESP8266 for the webinterface and interconnect the 2 through the serial interface.

GabeHC, 2017-02-02:
I suggest following changes.

1. make PREV NEXT to work on playlist on tracks when playing m3u
2. add random playing when in playlist mode
3. add m3u8 support, to do this will 
  - change html charset to utf-8
  - remove all ( b > 0x7F ) 
  - add 
```  
if ( host.endsWith ( ".m3u" )                     // Is it an m3u playlist?
    ||  host.endsWith ( ".m3u8" )                // or .m3u8 for UTF-8 playlist
   )
```

arlaor, 2017-02-08:
Hi @Edzelf 
You can support this radio station or this type of links
The two links belong to the same station and are supported by foobar2000 player
http://streams.deltaradio.de/delta-live/mp3-128/streams.deltaradio.de/play.m3u
http://streams.deltaradio.de/delta-live/mp3-128
The second link can be played from chrome

Edzelf, 2017-02-09:
The 2nd link yields a redirection to: regiocast.hoerradar.de/deltaradio128?sABC=589p85q8%233n79694784rn57rr01s34sp2s9305
Use this instead, as Esp-radio does not support redirection.

closed 2017-04-05 by GabeHC