Edzelf/Esp-radio #59: Random error while updating web page

arlaor, 2017-01-24:
Hi @Edzelf 
Sometimes happens that when you update the web page this does not reload the sp-radio page and sends the following code to the web page
After this does not reload the page, until you press the reset button

```javascript
     }
    }
    xhr.open ( "GET", theUrl, false ) ;
    xhr.send() ;
   }

   function handlepreset ( presctrl )
   {
    if ( presctrl.value >= 0 )
    {
      httpGet ( "preset=" + presctrl.value ) ;
    }
   }

   function handletone ( tonectrl )
   {
     var theUrl = "/?tone" + tonectrl.id + "=" + tonectrl.value +
                  "&version=" + Math.random() ;
     var xhr = new XMLHttpRequest() ;
     xhr.onreadystatechange = function() {
       if ( xhr.readyState == XMLHttpRequest.DONE )
       {
         resultstr.value = xhr.responseText ;
       }
     }
     xhr.open ( "GET", theUrl, false ) ;
     xhr.send() ;
   }

   function setstat()
   {
     var theUrl = "/?station=" + station.value + "&version=" + Math.random() ;
     var xhr = new XMLHttpRequest() ;
     xhr.onreadystatechange = function() {
       if ( xhr.readyState == XMLHttpRequest.DONE )
       {
         resultstr.value = xhr.responseText ;
       }
     }
     xhr.open ( "GET", theUrl, false ) ;
     xhr.send() ;
   }
   // Fill preset list initially
   //
   var i, select, opt, stations ;
   select = document.getElementById("selpres") ;
   var theUrl = "/?list=0" + "&version=" + Math.random() ;
   var xhr = new XMLHttpRequest() ;
   xhr.onreadystatechange = function() {
     if ( xhr.readyState == XMLHttpRequest.DONE )
     {
       stations = xhr.responseText.split ( "|" ) ;
       for ( i = 0 ; i < ( stations.length - 1 ) ; i++ )
       {
         opt = document.createElement( "OPTION" ) ;
         opt.value = stations[i].substring ( 0, 2 ) ;
         opt.text = stations[i].substring ( 2 ) ;
         select.add( opt ) ;
       }
     }
   }
   xhr.open ( "GET", theUrl, false ) ;
   xhr.send() ;
  </script>
  <script type="text/javascript">
    var stylesheet = document.createElement('link') ;
    stylesheet.href = 'radio.css' ;
    stylesheet.rel = 'stylesheet' ;
    stylesheet.type = 'text/css' ;
    document.getElementsByTagName('head')[0].appendChild(stylesheet) ;
  </script>
 </body>
</html>
<noscript>
  <link rel="stylesheet" href="radio.css">
  Sorry, ESP-radio does not work without JavaScript!
</noscript>
```

Edzelf, 2017-01-24:
Yes,  this is probably the case when there is no buffer space left for the TCP data.  I have the idea that it happens more when listening to mp3-files.  mp3 streams works better.  I will try to find a solution, but its not so easy.

GabeHC, 2017-01-28:
Adding spi dram will help?

Edzelf, 2017-01-28:
There is plenty of RAM in the ESP8266 for this project.  It's the lack of TCP-buffers in the the "BIOS" that has problem with multiple TCP connections.

arlaor, 2017-02-01:
When making a change directly in the CONFIG page, such as removing or increasing a radio station and saving the changes, with the SAVE and RESTART buttons, upon restarting the ESP-12E module can not be connected again until the file " Radio.ini "again via arduino IDE.
If more than 20 radio stations are added, the INDEX page is not displayed in the PRESET drop-down list and displays incorrect information on the "config" web page.

Edzelf, 2017-02-01:
The size of the radio.ini file is limited to about 3292 bytes, due to limited free memory space .  If you want more, you have to lower the size of the RINGBUF.
You can check the size of radio.ini by entering "testfile=radio.ini" in the serial monitor.
You may also uses TABs instead of multiple spaces in radio.ini and remove commentlines in order to preserve space.  This way I managed to have a list of 30 presets in my radio.ini (RINGUFSIZE=20000), 80 presets with RINGBUFSIZE=10000.
Tip: use STOP before changing the configuration.

arlaor, 2017-02-01:
Hi @Edzelf 
I have seen in some projects that use FS, they upload compressed files in ".gz", to gain space in memory.
Is there any possibility of doing the same with the file radio.ini, to be able to obtain a greater number of stations?

Edzelf, 2017-02-09:
radio.ini must be in a form readable bij the sketch.  For .gz file you need software to unzip the file in the ESP8266.

Birdstreet, 2017-12-13:
Decreasing the size of the .ini file helped in my case. And thanks for your job!