Edzelf/Esp-radio #60: SPIFFS files

fenyvesi, 2017-02-01:
I try to use a different VS1063b board (Sparkfun breakout board) without success: no sound. The board works, I tried it with Arduino.
For debugging I wanted to play a local mp3 file stored in SPIFFS. I put the mp3 files ( 2 * 11k) into the /data directory but they are not transferred in the process of uploading. I wasn't able to guess how to upload through the Web interface either.
Any advice?
Thanks.

fenyvesi, 2017-02-01:
Arduino v1.6.11 is used.

Edzelf, 2017-02-01:
I do not have a VS1063 board, so I can't test it.
For uploading to SPIFFS: press the "STOP" before uploading or saving configuration.

fenyvesi, 2017-02-01:
Thank you for the answer.

Sorry it was misspelled. VS1053B.
The board connects GPIO0, GPIO1, GPIO2, GPIO3 to Ground.  The Line_In, MIC_P to AGND through a capacitor. Reset, Rx to HIGH.

I can see the 1st chunk in the console, the status seems to be OK, but I can see often: 
Free memory is 10704, ringbuf 0, stream 0 (may be because of streaming and not having capacity for http.

The mp3 file i can play from browser doesn't work:

D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec
D: Command: station with parameter www.stephaniequinn.com/Music/Rondeau.mp3
D: Connect to new host www.stephaniequinn.com/Music/Rondeau.mp3
D: Connect to www.stephaniequinn.com on port 80, extension /Music/Rondeau.mp3
D: Connected to server
D: Content-Type: audio/mpeg
D: Last-Modified: Wed, 29 Jul 2009 16:21:00 GMT
D: Accept-Ranges: bytes
D: Server: Microsoft-IIS/8.5
D: X-Powered-By: ASP.NET
D: Content-Length: 2157423
D: Connection: close
D: Switch to DATA, bitrate is 0
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (6/0) from 205.164.62.15:10032

fenyvesi, 2017-02-01:
I tried to upload. It crashed.

: Command: stop (without parameter)
D: STOP requested
D: Stopping client
D: Song stopped correctly after 0 msec
D: Command: getnetworks with parameter 0
D: File upload gsm.mp3, len 1321, index 0
D: File upload gsm.mp3, len 308, index 10081
D: File upload gsm.mp3, 9068 bytes finished
D: FileRequest received /upload

Exception (28):
epc1=0x40207dbb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffbf0 end: 3fffffb0 offset: 01a0

>>>stack>>>

fenyvesi, 2017-02-01:
After restarting I have it:

D: FS Total 2949250, used 11546
D: /radio.ini                       -    1601
D: /gsm.mp3                     -    9068

The length is incorrect, the real length is 10.389 bytes instead of 9.068. In the uploading process seems to be a problem.

I can't play:

D: Command: stop (without parameter)
D: Command: station with parameter localhost/gsm.mp3
D: Switch to DATA, bitrate is 0
D: STOP requested
D: Song stopped correctly after 0 msec

Edzelf, 2017-02-01:
Even without a VS1053 connected the debug output should be like the text below.
The lines starting with "D: First chunk:" are missing.  That means that there is no data coming from the WiFi connection.  So first concentrate on that.  Do you have a steady WiFi connection?  Do you have a strong powersupply?
-----------------------------------------------------------------------------------------------

D: Connect to new host www.stephaniequinn.com/Music/Rondeau.mp3
D: Connect to www.stephaniequinn.com on port 80, extension /Music/Rondeau.mp3
D: Connected to server
D: Content-Type: audio/mpeg
D: Last-Modified: Wed, 29 Jul 2009 16:21:00 GMT
D: Accept-Ranges: bytes
D: Server: Microsoft-IIS/8.5
D: X-Powered-By: ASP.NET
D: Content-Length: 2157423
D: Connection: close
D: Switch to DATA, bitrate is 0
D: First chunk:
D: FF FB B0 60 00 00 00 00
D: 00 00 00 00 00 00 00 00
D: 00 00 00 00 00 00 00 00
D: 00 00 00 00 00 00 00 00

fenyvesi, 2017-02-01:
The ESP and the VS board powered by USB. Even 500 mA should be enough. The Wifi connection is stable, 1 meter from the AP.

If I connect a preset station, than tranfers data.

D: Song stopped correctly after 0 msec
D: New preset/file requested (4/0) from airspectrum.cdnstream1.com:8114/1648_128
D: Connect to new host airspectrum.cdnstream1.com:8114/1648_128
D: Connect to airspectrum.cdnstream1.com on port 8114, extension /1648_128
D: Connected to server
D: Content-Type: audio/mpeg
D: icy-br:128
D: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
D: icy-br:128
D: icy-description:Today's Easy Hits from the 70s, 80s, 90s through Today!
D: icy-genre:Easy Listening
D: icy-name:Easy Hits Florida
D: icy-private:0
D: icy-pub:1
D: icy-url:http://www.easyhitsflorida.com
D: Server: Icecast 2.4.0-kh3
D: Cache-Control: no-cache, no-store
D: Pragma: no-cache
D: Access-Control-Allow-Origin: *
D: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Connection: Close
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: icy-metaint:16000
D: Switch to DATA, bitrate is 128
D: Metadata block 48 bytes
D: Streamtitle found, 47 bytes
D: StreamTitle='Teddy Pendergrass - Love T.K.O. ';

fenyvesi, 2017-02-01:
But no chunk only title.

fenyvesi, 2017-02-01:
After restarting I always get the 1st chunk, no later. The LED is almost constantly lights. In the meantime the test: Free memory is 10720, ringbuf 0, stream 0

: New preset/file requested (-1/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
D: Server: Icecast 2.4.2
D: Content-Type: audio/mpeg
D: Cache-Control: no-cache
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Pragma: no-cache
D: icy-br:192
D: ice-audio-info: bitrate=192
D: icy-br:192
D: icy-genre:Talk
D: icy-name:Radio 1
D: icy-pub:0
D: icy-url:http://www.radio1.nl
D: icy-metaint:16000
D: Switch to DATA, bitrate is 192
D: First chunk:
D: F7 C2 DD 0C AF 1E 2A 44
D: 88 9E 3F 2D C4 2D 08 A5
D: 55 AF AE 42 5C 1B DD 3C
D: 88 90 57 2A 15 EB C6 12
D: Metadata block 64 bytes
D: Streamtitle found, 56 bytes
D: StreamTitle='NPO Radio 1 - Radio EenVandaag - AVROTROS';

fenyvesi, 2017-02-01:
I will try your latest version, now I use the previous one.

.

Edzelf, 2017-02-01:
I just fixed a bug concerning file upload: the first chunk was not saved to SPIFFS.
Ringbuf = 0 or stream = 0 means that incoming data is slower than it should be for the given bitrate.  If the led is on, it could mean that the VS1053 demands too much data.  Maybe DREQ is always low.  In that case you may have the line "VS1053 not properly installed!" in your debug output at startup.

fenyvesi, 2017-02-01:
I haven't seen the message but will check DREQ.
this is the debug after boot. The SCI registers are dumped twice.  Immediately before the chunk is printed, all the registers show 0.

D: FS Total 2949250, used 21837
D: /radio.ini                       -    1601
D: /gsm.mp3                         -    9068
D: /mqtt.mp3                        -    9851
D: Added SSID 00 = MADOX3G to acceptable networks
D: Added SSID 01 = TUSNAD5 to acceptable networks
D: * Scan Networks *
D: Number of available networks: 2
D:  1 - TUSNAD6                   Signal: -90 dBm Encryption WPA2  
D:  2 - MADOX3G                   Signal: -55 dBm Encryption WPA2  Acceptable
D: --------------------------------------
D: Command: mqttbroker (without parameter)
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: mqttpasswd with parameter none
D: Command: mqtttopic with parameter espradio
D: Command: mqttpubtopic with parameter espradioIP
D: Command: wifi_00 with parameter MADOX3G/Newton01
D: Command: wifi_01 with parameter TUSNAD5/Newton01
D: Command: volume with parameter 72
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: preset with parameter 6
D: Preset set to 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Starting ESP Version Tue, 23 Jan 2017 15:10:00 GMT...  Free memory 14648
D: Sketch size 340784, free size 704512
D: Reset VS1053...
D: End reset VS1053...
D: Slow SPI,Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: After last clocksetting
D: REG   Contents
D: ---   -----
D:   0 -  4800
D:   1 -    40
D:   2 -     0
D:   3 -  6000
D:   4 -     0
D:   5 -  AC44
D:   6 -     0
D:   7 -  1E06
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -  FFFC
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -     0
D: Selected network: MADOX3G                  
D: Try WiFi MADOX3G
D: IP = 192.168.0.201
D: Start server for commands
D: STOP requested
D: Song stopped correctly after 0 msec
D: New preset/file requested (-1/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
D: Server: Icecast 2.4.2
D: Content-Type: audio/mpeg
D: Cache-Control: no-cache
D: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Pragma: no-cache
D: icy-br:192
D: ice-audio-info: bitrate=192
D: icy-br:192
D: icy-genre:Talk
D: icy-name:Radio 1
D: icy-pub:0
D: icy-url:http://www.radio1.nl
D: icy-metaint:16000
D: Switch to DATA, bitrate is 192
D: first chunk
D: REG   Contents
D: ---   -----
D:   0 -     0
D:   1 -     0
D:   2 -     0
D:   3 -     0
D:   4 -     0
D:   5 -     0
D:   6 -     0
D:   7 -     0
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -     0
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -     0
D: First chunk:
D: DC E6 19 61 4B 01 4E CA
D: EC C8 ED 53 52 5C A6 D6
D: 3F FF AA 4E EA 9A 6E CC
D: 86 39 39 4D 6A 2D 7E 59
D: Metadata block 64 bytes
D: Streamtitle found, 56 bytes
D: StreamTitle='NPO Radio 1 - Radio EenVandaag - AVROTROS';

fenyvesi, 2017-02-01:
I compiled your latest version (30 minutes old) and I got the message you suggested. I check DREQ.

D: Reset VS1053...
D: End reset VS1053...
D: VS1053 not properly installed!

fenyvesi, 2017-02-01:
I'm not sure, but almost.

The problem is that after initialization th VS1053 is kept in reset. I was lazy and not having TFT I used GPIO15 for reset directly (without the diodes). But if TFT is not used you suppose that it is a button and goes low later so keeping the board in reset status forever.

I rearrange the setup tomorrow.

Thank you for your help and sorry for being so dumb.
George

fenyvesi, 2017-02-02:
It works.

closed 2017-02-02 by fenyvesi