Edzelf/Esp-radio #61: TFT Colours

thermo2016, 2017-02-05:
I have a problem with the colours shown on the TFT, they seem to completely messed up. 
I have a TFT which seems to match what is shown in the PDF, blue circuit board, "1.8 SPI 128x160" at the top, "1.8 TFT MODULE" at the bottom.
The colours are defined as:
#define	BLACK   0x0000
#define	BLUE    0xF800
#define	RED     0x001F
#define	GREEN   0x07E0
#define CYAN    GREEN | BLUE
#define MAGENTA RED | BLUE
#define YELLOW  0x07FF
#define WHITE   0xFFFF

But through trial & error I have found these are close:
#define BLACK   0xFFFF
#define CYAN    0x00E1
#define YELLOW  0x0070
#define WHITE   0x00E0

It appears the colour scheme on the TFT is not 5:6:5. Any ideas what is going on?

Edzelf, 2017-02-06:
The color scheme for my TFT is 5:6:5.  It seems strange that BLACK = 0xFFFF in your case.  Looks like the color bits are inverted.  So try the inversed values.
The library I used is TFT_ILI9163C-master version 0.9.
For the TFT module, I am using the 8 pins connector, the 16 pins on the other side are not used.

thermo2016, 2017-02-06:
I tried inverting the bits, but the colours didn't make sense. I think there is a hardware fault with the TFT. I have ordered replacements (red circuit board). I'll close when I can confirm.
BTW, I think your definition of BLUE & RED may be swapped, and YELLOW may be incorrect. From the TFT_ILI9163C-master example test.ino the colours are defined as:
#define	BLACK   0x0000
#define	BLUE    **0x001F**
#define	RED     **0xF800**
#define	GREEN   0x07E0
#define CYAN    0x07FF
#define MAGENTA 0xF81F
#define YELLOW  **0xFFE0**
#define WHITE   0xFFFF
My TFT shows these colours incorrectly as well.

thermo2016, 2017-02-16:
Replacement TFT has arrived, connected and tested. Using the default colours as defined, it works as expected.
I think the blue mother board TFT screens have quality issues. In my experience, the red motherboard TFT appears to work.
Issue closed. Problem was hardware fault.

closed 2017-02-16 by thermo2016