Edzelf/Esp-radio #62: Debug (and control?) via telnet

thermo2016, 2017-02-08:
I recently found this library:
https://github.com/JoaoLopesF/ESP8266-RemoteDebug-Telnet
I think it would be useful if esp-radio had debug messages via telnet. It could also be useful to control esp-radio via telnet in the same way it can be controlled via serial.

Edzelf, 2017-02-09:
Good idea.  See what I can do.

Edzelf, 2017-02-09:
Did not work.  Telnet commands stop the radio.  Maybe too much connections.

thermo2016, 2017-02-09:
Could it possibly be running out of memory? Could it be the stack is hitting the heap?
I tried the RemoteDebug-Telnet library, showing debug messages via telnet was no problem, but it would not run commands. Weird stuff was happening. Sometimes it would reset or crash on startup.
I reduced the size of the ringbuffer, it appears that it would then accept commands. Sometimes it would not display the full result of 'status' but it would do something.
I need to do more testing....

thermo2016, 2017-02-10:
I've done more testing. RINGBFSIZ=15000 and commented out/removed all MQTT stuff. Commands & debug via telnet work just fine.
Setting RINGBFSIZ=18000 causes problems. Command 'status' gives incomplete results, some other commands are ignored or give incomplete messages. 
Setting RINGBFSIZ=20000 (the default) causes even more problems. Most/all telnet commands are ignored. Esp-radio sometimes crashes on startup at random points at random times.

I've used the telnet library as above. I think indications are strong that there is a memory issue when setting RINGBFSIZ=20000. Could this be the reason for the problems with the web  interface?

I hope this helps.