Edzelf/Esp-radio #63: Preset not working

acocalypso, 2017-03-16:
Hi after playing around with it, I figured out that my preset is not laoding.

D: New preset/file requested (-1/0) from http://server1.blitz-stream.de:4400
D: Connect to new host http://server1.blitz-stream.de:4400
D: Connect to http on port 0, extension //server1.blitz-stream.de:4400
D: Request http://server1.blitz-stream.de:4400 failed!
D: No data input

radio.ini

# Presets
preset = 1					                                  # Start with preset 6
preset_00 = http://server1.blitz-stream.de:4400               #  0 - METAL-ONLY

But when I enter http://server1.blitz-stream.de:4400 and press play it is working.

Any idea what causes that issue?

Edzelf, 2017-03-16:
In the debug log you see that the portnumber derived from the preset is "0".  This should be "4400"
Note that the "http://" prefix is not allowed in the preset list in radio.ini.  Remove this prefix like:
preset_00 = server1.blitz-stream.de:4400 # 0 - METAL-ONLY
And it will work correctly.

acocalypso, 2017-03-16:
Thanks that worked for me :)

closed 2017-03-16 by acocalypso