Edzelf/Esp-radio #65: Compilation error.

folkos, 2017-04-04:
Hello! An error occurs during compilation: Arduino:
C: \ Users \ tol \ Desktop \ sketch_apr03a \ sketch_apr03a.ino: 294: 24: fatal error: about_html.h: No such file or directory

  #include "about_html.h"

                         ^

Compilation terminated.

Exit status 1

Help to solve the problem.

Edzelf, 2017-04-04:
Put the files "about_html.h", "config_html.h", "favicon.h" and "index_html.h" in the directory where your sketch is in.

folkos, 2017-04-04:
OK. In the evening I will try. Thanks for the advice.

folkos, 2017-04-04:
Hello ! I tried to compile on your advice, it still gives an error to another file. I add it to the folder with the sketch, it starts to swear already on another file. Could you make a sketch in expanding bin?

Edzelf, 2017-04-05:
Just give me the error message.

folkos, 2017-04-05:
Ok.I'll get to the computer, I'll send an error message.

folkos, 2017-04-05:
I added the files "about_html.h", "config_html.h", "favicon.h" and "index_html.h" to the folder with the sketch. Gives an error message :

C: \ Users \ C ... \ Documents \ Arduino \ sketch_apr03a \ sketch_apr03a.ino: 296: 23: fatal error: Arduino: 1.6.8 (Windows 7), Fee: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS) "

C: \ Users \ From ... Documents? Arduino \ sketch_apr03a \ sketch_apr03a.ino: 296: 23: fatal error: radio_css.h: No such file or directory

Compilation terminated.

Exit status 1

Compilation terminated.

I add a file with a sketch file radio_css.h  begins to swear on another file, and so to the infinite.

Edzelf, 2017-04-05:
So you added "radio_css.h" to the directory where your sketch is in.
What is the next error message?
Make also sure that you have the subdirectory "data" with (a modified) "radio.ini" in it.
Read the documentation in the .pdf file.

folkos, 2017-04-06:
Thank you very much for your help. I managed to download the sketch.

folkos, 2017-04-06:
Have you thought about using different displays? I hope for the further development of the project.
Thank you

Edzelf, 2017-04-06:
No, the display does the job.  But it should not be too difficult to replace it with a different display.