Edzelf/Esp-radio #66: Schematic and Analog buttons

hcremer, 2017-04-10:
Hello,

first of all, I would like to express my admiration and gratitude for the great job you have done programming this.
I first tried it with the Linux Arduino IDE, it compiles fine but it is not uploading the sketch.
I then switched to Windows and it worked.
I still have no sound, but I think my VS1053 is a piece of junk as it doesn't give any sound in the MIDI and MP3 tests with an arduino also (already ordered a new one).

I have problems understanding the circuit schematic on the reset circuit. I think the 5V and other wires are mixed up between the diodes and the VS1053. Could you maybe draw this more clear?

Another question is if it would be possible to add 2 more analog switches for volume up and down? This would be great for a standalone radio.

Again thanks a lot for this great piece of work.

Edzelf, 2017-04-10:
**Reset circuit.**
The reset circuit for the VS1053 is base on a OR-gate formed by the two diodes.  Only if both GPIO16 and GPIO05 are LOW, the VS1053 reset pin will go low.
I will correct the PNG-image.
But I think you can omit this circuit and just pull up XRST as the VS1053 does not need the reset in normal circumstances.
**Extra functions.**
I think it is possible to add 2 more analog switches.  A port expander is also possible.  And you may consider GPIO0 for a simple IR remote control.

hcremer, 2017-04-10:
Thank you very much for the quick response!

closed 2017-04-10 by hcremer