Edzelf/Esp-radio #67: how to solve Sluggish WEB UI response problem when playing MP3?

GabeHC, 2017-04-10:
You mentioned it is lack of TCP buffer, Do you figuring out how to?

Edzelf, 2017-04-10:
Some station deliver MP3 data faster than the decoder can handle according to the bitrate.  The TCP buffers will then be saturated, leaving no space for handling data for the web interface.  You may have noticed that some stations are working fine, while others have problems.
It would be nice if some buffer space could be reserved for the webinterface.
I also tried to speed up the bitrate of the decoder a little when the buffers are filling up, but it did not work in my case.  An other approach might be to skip frames in this case, but I haven't tried that.
A final solution is to use 2 a separate ESP for the webinterface....

Edzelf, 2017-05-23:
ESP32 version is released.  Performance seems much better with multiple TPC connections.

GabeHC, 2017-06-23:
You are amazing!
No more lost control while playing M3U MP3. Web interface response almost instantly. 
Thank you so much.

One Question thought: my VS1053 MP3 shield SD card mounting fails, it works fine on 5V arduinos. could it be logic level issues?

Edzelf, 2017-06-23:
Sounds like a wiring error.  Have you tried different SD cards?

jgrulich, 2018-01-10:
I've found that when disabled ArduinoOTA, it improves significantly the web response and overall stability.
BTW, HTTPUpdate may be better choice for OTA.