Edzelf/Esp-radio #68: Analog Input

hcremer, 2017-04-10:
I made some tests with analog input switches. My ESP-12F always reads 1024 when the voltage is over 1 V. So I needed to add a 15k resistor between 3.3V and the analog circuit.
Are other ESP's reading 0 - 3.3V with the same software?

Edzelf, 2017-04-11:
Yes, some models will measure 0-1 Volt, other 0-3.3V.  In your case you have to lower the resistor in the ladder that is connected to ground to about 2.7 kOhm.
You can use the web-interface or the serial interface to monitor the ADC reading.

hcremer, 2017-04-11:
Thank you very much for the quick answer!

NonaSuomy, 2017-05-08:
I did a simulation here [analog read 3.3v](http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.20027730826997+50+5+50%0Av+48+368+48+48+0+0+40+3.3+0+0+0.5%0Aw+48+48+96+48+1%0Aw+176+48+256+48+0%0Aw+256+48+336+48+0%0As+176+48+176+128+0+1+false%0As+256+48+256+128+0+1+false%0As+336+48+336+128+0+1+false%0Ar+96+48+96+112+0+10000%0Ar+96+240+96+304+0+10000%0Ar+96+304+96+368+0+10000%0Aw+96+176+256+128+0%0Aw+96+240+336+128+0%0Aw+96+304+192+304+2%0Aw+176+48+96+48+0%0Ar+96+112+96+176+0+10000%0Ar+96+176+96+240+0+10000%0Aw+48+368+96+368+0%0Aw+96+112+176+128+0%0A)

[analog read 1v](http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.20027730826997+50+5+50%0Av+48+368+48+48+0+0+40+3.3+0+0+0.5%0Aw+48+48+96+48+1%0Aw+176+48+256+48+0%0Aw+256+48+336+48+0%0As+176+48+176+128+0+1+false%0As+256+48+256+128+0+1+false%0As+336+48+336+128+0+1+false%0Ar+96+48+96+112+0+10000%0Ar+96+240+96+304+0+10000%0Ar+96+304+96+368+0+10000%0Aw+96+176+256+128+0%0Aw+96+240+336+128+0%0Aw+96+304+192+304+2%0Aw+176+48+96+48+0%0Ar+96+112+96+176+0+10000%0Ar+96+176+96+240+0+10000%0Aw+48+368+96+368+0%0Aw+96+112+176+128+0%0A)

If you change R5 to anything under 4.3k you will be in the 1v range.

NonaSuomy, 2017-05-09:
Any chance @Edzelf can add something to the documentation about this as a lot are 1v analog read.

Edzelf, 2017-05-09:
Fixed.

closed 2017-06-08 by hcremer