Edzelf/Esp-radio #69: SPI problem with ESP32

madjid13, 2017-04-12:
hello,

I have a problem with my esp32 i have bought to aliexpress:
[image]
[image]

When i connect it to my sd card spi interface, i have this message in my serial windows arduino:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
cmd len -1
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
cmd len -1
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
cmd len -1
Falling back to built-in command interpreter.
OK
Could helpppp me, thanks!!!!

Edzelf, 2017-04-13:
Sorry, I do not own any ESP32.  I will order one from Ali.
The ESP-radio software is designed for ESP-12.

andybrk, 2017-04-27:
I did check the SW (with modifications) under ESP32. (without Web interface yet)
I move the SPIFFS to the SD card and this has absolutely no problems.  May be you have a hardware issue.
The problem with the ESP32 is different:
Have to modify the timer (no problem)
All seems to work, but no sound (I suppose the ESP32 is more slowly under Arduino, but I'm still investigate)
The WIFI and web functions are complete different (should not be a problem but hard work)

madjid13, 2017-04-27:
Thanks a lot
I will try again.
Have a nice day.

Edzelf, 2017-04-27:
Got my ESP32 now.  I'm working on the ESP32 version.  Fixing the webinterface is quite a job.

andybrk, 2017-04-27:
The speed is real a problem, too. I did measure this sub with a scope:  putring ( mp3client.read() ) ;
need for esp8266 7us and for the esp32 80us about (include digitalWrite). For the esp32 is missing the "user_interface.h". This could be the reason that the SW can't use direct the espressif libraries

madjid13, 2017-04-27:
Edzelf:please if you have the same esp32 iot from Aliexpress, could tell me exactly how sdcard

madjid13, 2017-04-27:
Is working and share schematics pinout.
Thanks 🤔

Edzelf, 2017-05-23:
ESP32 version is released!