Edzelf/Esp-radio #71: Can't send URLs with symbols "&" tried &amp; and %26 URL gets cut off.

NonaSuomy, 2017-04-24:
Example URL http://10.13.37.65/?station=http://api.voicerss.org/?f=32khz_16bit_stereo&key=29334415954d491b85535df4eb4dd821&hl=en-us&src=hello?the-sun-is-warm-and-i-like-it

Console shows it got the url: http://api.voicerss.org/?f=32khz_16bit_stereo
I removed the & symbol and then it shows:
http://api.voicerss.org/?f=32khz_16bit_stereokey=29334415954d491b85535df4eb4dd821
etc... I tried to use url escape codes but they have symbols in them that it doesn't like as well so they get cut off, ie &amp; and %26.

Also tried to send a URL with & in it from the website send box and it did the same thing, I will test the console commands next.

Edzelf, 2017-04-24:
Your "station"-command is probably too long (159 characters).  Try to increase the size of "cmd" from 130 to 200 and the value of "DEBUG_BUFFER_SIZE" to 200.
Note that there is no need to include "http://" in the station command (saves another 7 bytes..).

NonaSuomy, 2017-04-24:
I'll give that a shot increasing those values, does it start counting after = or station= included or is the entire URL included?

```
http://10.13.38.65/?station=
?station=http://api.voicerss.org/?f=32khz_16bit_stereo&key=29334415954d491b85535df4eb4dd821&hl=en-us&src=hello-the-sun-is-warm-i-like-it
```

Equals 155 Characters: [letter counter](http://www.lettercount.com/)

Also 
```
station=http://api.voicerss.org/?f=32khz_16bit_stereo&key=29334415954d491b85535df4eb4dd821&hl=en-us&src=hello-the-sun-is-warm-i
```
Works from the serial interface but 
```
http://10.13.38.65/?station=
?station=http://api.voicerss.org/?f=32khz_16bit_stereo&key=29334415954d491b85535df4eb4dd821&hl=en-us&src=hello
```
Count: 139 (Full URL)

Doesn't work from the web browser, will increasing that debug_buffer_size help that?

Maybe there is not enough memory on the ESP8266 to handle this but valid URLs can be as long as 2000 characters give or take for most standard web browsers.

Excerpt from (SOF)[http://stackoverflow.com/questions/417142/what-is-the-maximum-length-of-a-url-in-different-browsers]

[RFC 2616](http://www.faqs.org/rfcs/rfc2616.html) (Hypertext Transfer Protocol HTTP/1.1) section 3.2.1 says
```
The HTTP protocol does not place any a priori limit on the length of a URI. Servers MUST be able to handle the URI of any resource they serve, and SHOULD be able to handle URIs of unbounded length if they provide GET-based forms that could generate such URIs. A server SHOULD return 414 (Request-URI Too Long) status if a URI is longer than the server can handle (see section 10.4.15).
```

That RFC has been obsoleted by [RFC7230](http://tools.ietf.org/html/rfc7230#section-3.1.1) which is a refresh of the HTTP/1.1 specification. It contains similar language, but also goes on to suggest this:

```
Various ad hoc limitations on request-line length are found in practice. It is RECOMMENDED that all HTTP senders and recipients support, at a minimum, request-line lengths of 8000 octets.
```

NonaSuomy, 2017-04-24:
Not sure if this helps any,

<Rickta59> https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L244
<Rickta59> that seems to be where it starts parsing the args
<shbrngdo> I'm just anticipating the usual problems that show up on a) microcontroller b) trying to be a web server c) the way networking behaves sometimes when you can't have large buffers
<Rickta59> this seems to be where it parse out each arg ... I don't see any limit https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebRequest.cpp#L206
<shbrngdo> how much RAM is available on that device (esp8266) anyway
<Rickta59> lots
<NoSu> Rickta59, in the esp-radio ino I think host only holds the first part of the URL before /
<shbrngdo> well, lots = megabytes, or ?
<Rickta59> 10s of k
<shbrngdo> ok, so a URL line should be "bufferable" before parsing it.  I assume it's re-assembled from the MTU 'chunks' properly?
<NoSu> 45k
<Rickta59> I would put some debug prints in the _parseReqHeader routine and see if you are receiving the entire string
<shbrngdo> ^^^ yes
<Rickta59> if that is true .. then keep dive deeper and deeper until you find where it is chopped
<Rickta59> from what I see here it seems like there is no limit
<Rickta59> you might send a bogus arg after the station= ...
<Rickta59> something like &argends=1
<Rickta59> so the parse routine can see the end of your tts string
<shbrngdo> the other thing to consider, when an HTTP request is sent, it's a whole boatload of crap to wade through
<Rickta59> he is just looking at a get request  ...
<shbrngdo> you could easily exceed an MTU with the first chunk, which might cause your URLs to be truncated
<Rickta59> the info he needs is on the first line
<shbrngdo> right it's a GET request but it is HTTP/1.1 ?
<Rickta59>  I don't know how the esp webserver works
<shbrngdo> use nc to listen on a port, and do a simple GET request from firefox to your nc application.  watch the output.
<Rickta59> he is using a web browser to send the url
<shbrngdo> I'm anticipating that the web browser's request has a boatload of cruft in it, which may exceed the MTU, causing the request to be truncated if it's not re-assembled into a buffer
<shbrngdo> or it might exceed a buffer size that's too small
<shbrngdo> one nice test would be to send an HTTP/1.0 request, by hand, using nc or similar utility
<Rickta59> do you have any long real mp3 stations NoSu ?
<Rickta59> are they limited to 130 bytes?
<Rickta59> or is it just the ip address of the shoutcast station?
<NoSu> that's a hard one as most stations are just a url:port/mp3
<NoSu> most people don't have massivedomainnamegarbledygook123408230948etc.com domain names
<Rickta59> right ..
<Rickta59> https://github.com/Edzelf/Esp-radio/blob/master/Esp_radio.ino#L2476
<Rickta59> that does seem to limit the length of the string
<Rickta59> you are co-opting the station thing ...
<Rickta59> might be better to deal with it differently
<Rickta59> as it assumes you are only going to be using a short string
<Rickta59> make it "speakmsg" or something ...
<Rickta59> then implement it similar but use different buffers
<shbrngdo> I did a quick check - entering the example URL from earlier, I got a 492 byte HTTP/1.1 request.  but that's on my machine.  not sure what other browsers might do.  But the URL was at the beginning, so overrun might not be relevant.
<Rickta59> https://github.com/Edzelf/Esp-radio/blob/master/Esp_radio.ino#L2583
<Rickta59> seems like that line, you could corrupt memory if you send a really long string
<shbrngdo> ouch - nice find then
<Rickta59> and then it just be acting weird not because it truncates the get arg .. but because memory has become corrupt
<shbrngdo> exactly - buffer too small?
<shbrngdo> I hadn't considered buffer overrun [assuming that much was obvious... guess not]
<NoSu> do you mean if you comment out that sprintf statement it might just work with more chars
<NoSu> characters
<NoSu> as it did crash one time when I sent it a long URL
<NoSu> then when it rebooted I sent it again and it played.
<shbrngdo> that would make sense, yeah.  sprintf if using '%s' and similar is likely to overrun with unexpected data.  'snprintf' helps

NonaSuomy, 2017-04-25:
so tried with DEBUG_BUFFER_SIZE set to 200 still works from the console to a max of 
```
D: Connect to new host api.voicerss.org/?f=32khz_16bit_stereo&key=29334415954d491b85535df4eb4dd821&hl=en-us&src=hello-this-is-that-extended-par
D: Connect to api.voicerss.org on port 80, extension /?f=32khz_16bit_stereo&key=29334415954d491b85535df4eb4dd821&hl=en-us&src=hello-this-is-that-extended-par
```
Character Count 128

Basically the same as before, from the serial console, and through the web interface it still doesn't work.

NonaSuomy, 2017-04-25:
Got it working with _POST instead of the GET url, it definatly trunicates the url because of the ampersand as first questioned.

**SEND FROM _POST** 
```
curl --data "station=api.voicerss.org/?f=32khz_16bit_stereo%26key=29334415954d491b85535df4eb4dd821%26hl=en-us%26src=hello-this-is-that-extended-part-two-woot-woot-chaboogie" 10.13.38.65
```

Post as well first didn't work and cut off the URL at the ampersand, but the one thing post allowed was escaping the ampersands with %26 which using the web URL didn't.

NonaSuomy, 2017-04-27:
This is what I got working so far:
[OpenHab](http://www.openhab.org) is Home Automation Software
[ESP-Radio with Text To Speech for OpenHab](https://gist.github.com/NonaSuomy/8a41816defe7b3d918c147d9d274c101)
Next I may try to get the stations changing channels through OpenHab.