Edzelf/Esp-radio #72: set volume for each main loop pass?

andybrk, 2017-04-24:
In the actual SW will change the volume each main loop pass.
I think its better to do it so to lost not so much of time:

  bool           old_muteflag;
......
  if (!( muteflag == old_muteflag )) {
    old_muteflag = muteflag;
    if ( muteflag ) {                                   // mute / unmute 
      vs1053player.setVolume ( 0 ) ;                    // Mute
    } else {   
      vs1053player.setVolume ( ini_block.reqvol ) ;     // Unmute
    }
  }

Edzelf, 2017-04-24:
The requested volume will be written to the VS1053 register only if there is a change.  If nothing chnges, setVolume returns without any action.
Check the code of "setVolume".

andybrk, 2017-04-24:
Ok, this I didn't see. But you save the sub routine call if you catch this case in the main. I don't know which and how many registers will safe the ESP SW for a call. I did not analyse the timing if this could be  important or not. I suppose the ESP has much to do. ;-)
Just I try understand the SW to make this SW for the ESP32.

Edzelf, 2017-04-24:
Note that "volume"-, "upvolume"- and "downvolume"-commands will not work anymore with your code.
The function call will probably take less than 0.5 microseconds and even a delay of several milliseconds will not harm in the mail loop.
I will get my ESP32 next week (I hope) and I will try to make e Esp-radio for it.

andybrk, 2017-04-26:
You are right. This I didn't see.
I had in my mind that we need 50us about to put one byte to the VS1053 for higher bit rate. But the filling of the VBS103 is running in the both first wile loops of the main loop. So we have much of time. ;-)
If you can make this SW for the ESP32 this would be fine. My planning is to use a SD card for the mp3 files, the configuration and the websites. So we can have a fine mpeg player. I want t use this with a touch screen.
The other idea is, make a mod for the ESP8266 and remove GPIO9 and 10 from the flash. Then we have enough GPIOs for an additional SD card. (use the same SPI) but not for a touch screen.

closed 2017-04-26 by andybrk