Edzelf/Esp-radio #73: Get Current Station ID

Sander0542, 2017-04-25:
Hi,

I am the developer of the ESP Radio App, and I would love to have a option to get the current preset id.

Regards,
Sander

Sander0542, 2017-04-25:
or when you do this:

`http://192.168.178.24/?uppreset=1`

that it says `current preset is 3` instead of `command accepted`

Edzelf, 2017-04-26:
Fixed.  Output will look like: "Preset is now 6"

Sander0542, 2017-04-27:
Thanks you so much!

This is very hard needed for the tablet app

Sander0542, 2017-04-30:
I just installed the update, but when you are on preset 0 and go prev I go to -1 and then -2

Edzelf, 2017-05-01:
If the preset is out of range (like -1), the software will detect "no data" and will switch to the next higher preset.
I realize that the selected preset is not reflected to the app automatically.

NonaSuomy, 2017-05-13:
ok will do, I cleaned it up a bit, and then found out it sends blank values just fine :+1: also changed it so it didn't retain and had QoS of 0 in case those things were causing the connection drop and no luck still dropped the connection.

Here's the  refactored code: [GIST](https://gist.github.com/NonaSuomy/e50cfdf079fe1bbbcac30a381e34fd93)

[image]