Edzelf/Esp-radio #74: Request: First time boot with no radio.ini start up in wireless ad-hoc mode.

NonaSuomy, 2017-04-27:
Any chance you could make so the esp8266 starts up in ad-hoc mode if there's no radio.ini then the person could configure their AP then save a radio.ini for next boot.

Edzelf, 2017-04-28:
Esp-radio was designed to start up in AP mode if no ini-file is found.  I must admit I did not test this thoroughly.

NonaSuomy, 2017-04-28:
Ok I will test again, I somehow got stuck where it spammed the serial console with messages that it couldn't find the ini, but once I figured out how to upload the data folder with the arduino ide it started to work so didn't try it again.

NonaSuomy, 2017-05-02:
Ok so I was able to see the AP, but when I connect to Esp-radio with the password Esp-radio it connects and then when I try to go to the url in the browser 192.168.4.1 I html source code:

```
q.statusText ;
          }
        }
        oReq.send ( oData ) ;
        return false ;
      }

      // Fill configuration initially
      // First the available WiFi networks
      var i, select, opt, networks, params ;

      select = document.getElementById("selnet") ;
      var theUrl = "/?getnetworks=0" + "&version=" + Math.random() ;
      var xhr = new XMLHttpRequest() ;
      xhr.onreadystatechange = function() {
        if ( xhr.readyState == XMLHttpRequest.DONE )
        {
          networks = xhr.responseText.split ( "|" ) ;
          
          for ( i = 0 ; i < ( networks.length - 1 ) ; i++ )
          {
            opt = document.createElement( "OPTION" ) ;
            opt.value = i ;
            opt.text = networks[i] ;
            ssid.add( opt ) ;
          }
        }
      }
      xhr.open ( "GET", theUrl, false ) ;
      xhr.send() ;

      // Now get the configuration parameters from radio.ini
      theUrl = "/radio.ini" ;
      xhr.onreadystatechange = function() {
        if ( xhr.readyState == XMLHttpRequest.DONE )
        {
          inifile.value = xhr.responseText ;
        }
      }
      xhr.open ( "GET", theUrl, false ) ;
      xhr.send() ;
    </script>
    <script type="text/javascript">
      var stylesheet = document.createElement('link') ;
      stylesheet.href = 'radio.css' ;
      stylesheet.rel = 'stylesheet' ;
      stylesheet.type = 'text/css' ;
      document.getElementsByTagName('head')[0].appendChild(stylesheet) ;
    </script>
  </body>
</html>
```

Edzelf, 2017-05-03:
Fixed: New version will no longer try to connect if no Internet connection.
The fact that you see the source code cannot reproduced here.
Check your config_html.h file.

NonaSuomy, 2017-05-09:
Thanks, nothing was wrong with config_html.h probably had something to do with corrupt SPIFFS or memory overflow. After reflashing SPIFFS it has yet to dump the code, I have yet to have it dump the code again. Have yet to get AP properly working though, will keep looking and report back after playing with other items.

Edzelf, 2017-05-10:
Yes, SPIFFS is not very reliable.  For that reason there is a "testfile" function available in the radio.  Specially for long files it will show read problems.

robpan3, 2017-11-14:
I´m sorry, but I don´t understand where the file "radio.ini" man must put.
The error on "serial monitor" is "File /radio.ini not found!"
Thanks a lot for any help.
Roberto
Italy

Edzelf, 2017-11-14:
Roberto, the radio.ini file must be transferred to the SPIFFS.  It should be in your "data" subdirectory an can be uploaded by the IDE,  menu tools-> ESP8266 Sketch Data Upload.

robpan3, 2017-11-14:
Thanks a lot, ED for your suggestions.
I have read with "attention" your PDF of the project.
I have create a folder "Tools" instead "tools".  I think is was the problem.
Now I understand someting more about SPIFFS. 
Thanks very much for your speed answering.
Your project is fantastic.  I hope (I'm sure)  to complete all.
Thanks.
Roberto
Aosta
Italy

robpan3, 2017-11-14:
"ESP8266 Sketch Data Upload" on Arduino IDE 
Perfect!  Thanks!

Edzelf, 2017-11-14:
I meant the Tools option/tab in the IDE, not a folder.....

robpan3, 2017-11-15:
Again a little question.
You said that you use **14148** Diodes (in the page https://github.com/edzelf/esp-radio/issues/8). I think you use **1N4148**.  
But can I use 1N4007 instead? I already have it at home.
I 'm already able to receive my preferred station (without audio), only on serial monitor. 
I'm already satisfied for the moment.  But I am beginner! I'm studiing also the SPIFFS.
Excuse me for my poor english.
Thanks again!
And again many many compliments for your complex and big big project!
Roberto 
Italy

Edzelf, 2017-11-15:
The reset circuit is not really necessary.  Just connect the reset line of the VS1053 to the reset line of the ESP6288.
If you like, any diode will do, but keep the lines short, as the radio signals of the wifi may cause a DC signal on the diodes.

robpan3, 2017-11-16:
Thanks, Ed!

atacanteknik, 2017-11-19:
i add diodes  on vs1053 its very simple

robpan3, 2017-12-14:
Hallo Ed, 
I have almost realized the Espradio without the display. But it RUN!!!! You have made a SUPER SUPER Project! 
I don't have time for hobby because my job!! 
I don't have the precise TFT display of the project. 
Can I use the Nokia 5110 instead? 
If not, I wait from China the correct component. 
Thanks a lot! 
How can I contribute for your project/projects? 
I'm very satisfied!! 
Roberto 
Aosta 
Italy

Edzelf, 2017-12-14:
The 5110 is very limited.  It is worth the trouble of waiting for a better display.  Order a ESP32 as well, because it is much better than the 8266.

Birdstreet, 2018-01-25:
Compressing the .html  .css  and javascript part of the .html files of the web UI make the radio work more stable in AP mode. No partly loaded content issue.

olavsson, 2018-02-16:
How do you compress the .html .css and javascript part of the .html files?

Edzelf, 2018-02-16:
@Roberto. A small donation on my paypal account (ed@smallenburg.nl) will be welcome of course.

msafu2668, 2019-07-15:
hi Ed..I am first here.. where to put the data folder..?

Edzelf, 2019-07-16:
In the same directory where the sketch (the .ino file) is located.

bnicolae86, 2023-07-13:
I cannot make it work. The `data `folder is located on the same directory but I still receive this message: `File /radio.ini not found!
` . Maybe that's why the wifi connection is not working. I modified `radio.ini `  and put it inside data folder.

Edzelf, 2023-07-16:
Yes, the file radio.ini should be in the data folder.

bnicolae86, 2023-07-16:
> Yes, the file radio.ini should be in the data folder.

I wrote a sketch to read the file from SPIFFS and the file radio.ini is present,  I can view its content. But when I upload esp radio sketch I receive this message, radio.ini file not found. Maybe is something that I'm doing wrong.

Edzelf, 2023-07-16:
Please show the error message and the surrounding texts.

bnicolae86, 2023-07-17:
> Please show the error message and the surrounding texts.

After sketch upload I push also the radio.ini file

```
[SPIFFS] data    : C:\Users\user\Downloads\esp-radio\Esp_radio\data
[SPIFFS] size    : 3048
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/radio.ini
[SPIFFS] upload  : C:\Users\user\AppData\Local\Temp\arduino_build_930043/Esp_radio.spiffs.bin
[SPIFFS] address  : 0x100000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : COM3
[SPIFFS] speed    : 115200
........
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 3121152 bytes to 5246...
Writing at 0x00100000... (100 %)
Wrote 3121152 bytes (5246 compressed) at 0x00100000 in 0.5 seconds (effective 53831.0 kbit/s)...
Hash of data verified.
```

radio.ini file cannot be found 
```
D: FS Total 3121152, used 16384
D: File /radio.ini not found!
D: * Scan Networks *
D: Number of available networks: 6
...
D: File /radio.ini not found, use save command to create one!
D: Starting ESP Version Fri, 11 Feb 2022 13:05:00 GMT...  Free memory 22880
D: Sketch size 443120, free size 602112
D: Reset VS1053...
D: End reset VS1053...

Edzelf, 2023-07-17:
Ther is something wrong with the file upload.  You should see something like:
```
D: FS Total 1024000, used 24576
D: radio.ini                        -    1765
```
If you can connect using the network "Esp-radio" at 192.168.4.1, you may paste the contents of radio.ini in the config page, edit it and save it.

bnicolae86, 2023-07-17:
> Ther is something wrong with the file upload. You should see something like:
>
> ```
> D: FS Total 1024000, used 24576
> D: radio.ini                        -    1765
> ```
>
> If you can connect using the [...]

I tried that but my ESP keeps rebooting with a message like this:
``` 
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

>>>stack>>>

<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------