Edzelf/Esp-radio #75: Request: Optional SRAM instead of ring buffer

NonaSuomy, 2017-04-27:
[23LC1024](http://www.microchip.com/wwwproducts/en/23LC1024)
[23LCV1024](http://www.microchip.com/wwwproducts/en/23LCV1024)
[PDF 23LC1024](http://ww1.microchip.com/downloads/en/DeviceDoc/20005142C.pdf)
[PDF 23LCV1024](http://ww1.microchip.com/downloads/en/DeviceDoc/25156A.pdf)

I've tested samples of these parts from Ti with Ka-Radio and they work.

23LC1024 has Vhold <== using this one.
23LCV1024 has VBAT

*Note: If your chip has a /Vhold in place of /VBAT, the pin 7 must be wired to VCC (pin8)* instead of pin 7 to GND for the V series..
*Note2: Pin3 is not used in SPI mode and should not be left floating on the 23LC1024 (23LCV1024 pin3 is N/C).

1:CS = GPIO16 
2:MISO = GPIO12 
3:NC = VCC 3.3v
4:GND 
5:MOSI = GPIO13 
6:SCK = GPIO14 
7:VCC 3.3v 
8:VCC 3.3v

andybrk, 2017-04-27:
For what? I think the internal ram is much more faster than a ram on the SPI interface. We  have to transfer one byte for max bitrate  (320kb) in 25us about.

Edzelf, 2017-04-27:
Additional RAM will not have any effect on the performance of Esp-radio.  The firmware will still use the limited TCP buffers.

NonaSuomy, 2017-04-27:
What are the TCP buffers limited to?

Edzelf, 2017-04-27:
The buffers are allocated by the Espressive firmware.  There's no way to add extra buffers.
So when the mp3-stream fills up all available buffers, communication with the webinterface and/or mqtt will be blocked.  Some stations send more data than the VS1053 can handle.  This causes the TCP input-buffers to saturate.
I'm working on the ESP32 version now.  Maybe this one has a better schema for using the TCP input buffers.

NonaSuomy, 2017-04-27:
Thanks for the explanation, well I hope you still work on the esp8266 version as it has the lowest entry point currently and seems to work well so far, also glad you did it with Arduino, do you happen to know why when trying to play a small sample of say 1 second or few seconds around that it starts of choppy in the audio? When I tell it to stream the Text To Speech files it sometimes chops the first few words up a bit. Any way you can think to prevent it?

h1aji, 2021-05-24:
Hi @Edzelf 
I found this project [ESP8266Audio](https://github.com/earlephilhower/ESP8266Audio ) where SRAM is being used. 
Can you tell what is done differently there? Is there any chance to use that approach for Esp-radio?
Thanks

Edzelf, 2021-05-24:
I don't know what the problem is with your radio, but I think extra memory will not help to fix it.  In fact, the ringbuffer can be extended, as it is not using all of the available memory.

h1aji, 2021-05-25:
The issue is interruptions when the radio works. Only low bitrates works fine, like 64. I checked the antenna and it seems fine.
I have seen also on Karadio project optional SRAM which improves this situation

Edzelf, 2021-05-25:
I do not think the lack of memory is the problem.  Please give me the station that is causing the problem, so I can try to reproduce the error.

h1aji, 2021-06-05:
for example this one: http://orion.shoutca.st:8554/stream

> D: Connect to orion.shoutca.st on port 8554, extension /stream
> D: Connected to server
> D: Switch to HEADER
> D: Accept-Ranges:none
> D: Access-Control-Allow-Origin:*
> D: Cache-Control:no-cache,no-store,must-revalidate,max-age=0
> D: Connection:close
> D: icy-name:DI Radio Digital Impulse - Alex Stendor The Space Trace
> D: icy-genre:Electronic
> D: icy-br:320
> D: icy-sr:44100
> D: icy-url:http://www.di-radio.com
> D: icy-pub:1
> D: content-type:audio/mpeg
> D: udio/mpeg seen.
> D: icy-metaint:8192
> D: X-Clacks-Overhead:GNU Terry Pratchett
> D: Switch to DATA, bitrate is 320, metaint is 8192
> D: First chunk:
> D: FF FB E2 44 F9 82 07 A4
> D: 71 54 4B 8F 4F 80 D2 CD
> D: 1B 4F 6B 0C 2D 22 51 DD
> D: 4C 8E 61 93 CC 2B BD 6A
> D: Metadata block 64 bytes
> D: Streamtitle found, 63 bytes
> D: StreamTitle='Alex Stendor - The Space Trace Mix vol 16 [20N5]';

Edzelf, 2021-06-07:
I was able to reproduce the problem.   I guess 320 kbps is to much for the processor.  The ESP32 version of the radio will play this station without problems.

h1aji, 2021-06-07:
Would it be possible to add support for 23LC1024 SPI RAM? like descirebed in this project [ESP8266Audio](https://github.com/earlephilhower/ESP8266Audio) 

> Using external SPI RAM to increase buffer
A class allows you to use a 23lc1024 SPI RAM from Microchip as input buffer. This chip connects to ESP8266 HSPI port and provides a large buffer to help avoid hiccus in playback of web streams.

I have got this SRAM chip and keen to test. Say if it could be added as USESRAM option.

I created fork and modified your code slightly to work with LCD 2004 and IR receiver
[image]

Edzelf, 2021-06-07:
As I said, the memory is probably not the problem.  And I dont think SPI memory will be faster than internal RAM.

h1aji, 2021-06-13:
What is the maximum value for RINGBFSIZ ?

>  And I dont think SPI memory will be faster than internal RAM.

In this case question is not speed but buffer size, having an extra 1Mb buffer should store 3 seconds if you play 320kbits/s mp3 stream. You can see it in Ka-radio project (also in ESP8266Audio), and people reported (on russian forum) that it fixed issues with hiccups.

Edzelf, 2021-06-13:
The TEST button in the webinterface will tell you the amount of free memory.  Increase the ring buffer size accordingly.

h1aji, 2021-07-12:
@NonaSuomy would you be able assist with testing code for spi ram?