Edzelf/Esp-radio #76: EarSpeaker spatial processing

Sindi1, 2017-04-28:
Hi, 

first of all - thank for the great project!

Sound is great while using headphones. But using an amplifier with speakers 4 meters away from each other stereo sound is not very good. Compared to my web radio on the phone I was a little bit disappointed. 
Checking the data sheet  for the VS1053 I think I found the reason: 
"For the best headphone listening experience the VS1053 includes EarSpeaker spatial processing which accurately simulates how a room with stereo loudspeakers would sound. This option can naturally be turned off when required."

So my question is: Is there an option to turn this feature on / off? Would be great!

Second question: Has anybody a solution for running ESP / VS1053 and an amplifier with grounded audio input on same power ground? 

Regards,
Bernd

andybrk, 2017-04-28:
The problem is that the VS1053 have not GND related outputs. I think the best solution is a preamplifier with differential inputs or use a small NF transformer betwen the VS1053 and your amplifier.
In default (see 3.6.1) is the spartial prcessing is on (default 0x4800). But the SW do switch off that in line 568. ( write_register ( SCI_MODE, _BV ( SM_SDINEW ) | _BV ( SM_LINE1 ) ) ; in VS1053::begin() )

Sindi1, 2017-04-29:
Tahnks. So how should the line look like to turn spartial prcessing on?

andybrk, 2017-04-29:
First has to define first the new used bits: (add after line 333)
const uint8_t SM_EARSPEAKER_LO          = 4 ;        // Bitnumber in SCI_MODE for spatial processing L
const uint8_t SM_EARSPEAKER_HI           = 7 ;        // Bitnumber in SCI_MODE for spatial processing H

Then in line 469 ( 3 possibilities)
minimal spatial processing:
( write_register ( SCI_MODE, _BV ( SM_SDINEW ) | _BV ( SM_LINE1 ) | _BV (SM_EARSPEAKER_LO) ) ;
normal spatial processing:
( write_register ( SCI_MODE, _BV ( SM_SDINEW ) | _BV ( SM_LINE1 ) | _BV (SM_EARSPEAKER_HI)  ) ;
extreme spatial processing:
 ( write_register ( SCI_MODE, _BV ( SM_SDINEW ) | _BV ( SM_LINE1 ) | _BV (SM_EARSPEAKER_LO) | _BV (SM_EARSPEAKER_HI ) ) ;

But I doesn't think that this is your problem.

Edzelf, 2017-04-29:
Sounds like this is a problem of the amplifier.  Note that the VS1053 has no grounded common for the output signals.  So you have to supply a separate power supply or use audio  transformers, 40 cents a piece on aliexpress.

Sindi1, 2017-04-29:
Thanks. I know that VS1035 has no grounded common. For now I am using separated power supplies.  
I thought there might be an easy way like putting a capacitor between the output common and the amplifiers grounded input. 

First  "problem" was the (stereo) sound compared to another webradio / source. Sounds a little bit like one reverse polarity speakers and I thought this might be the spatial processing. The headphone sound was so good, the amplifier / speaker combination also when using a different source. 

So I will have to make some more comparison.

Sindi1, 2017-05-01:
I was a problem of the speakers and the distance between them. And the basement is no a good place to do a sound check.

Kestranor, 2017-06-12:
Sorry to hijack this thread, but I have a sound problem when connecting to a class D amplifier as well, even using separated power supplies it comes out very distorted and seems to be somewhat slowed as well. Is there a recommended way to connect the jack output from the VS1053 to a class D amplifier?
Thanks.

Edzelf, 2017-06-12:
Make sure that it works correctly with earphones.  Then make sure that the ground of the VS1053 is NOT connected to the ground of the amplifier or its power supply.  Th "middle" connection of the analog output of the VS1053 must be connected to the ground of the amplifier an its power supply.