Edzelf/Esp-radio #78: Request: Convert esp-radio.pdf => README.md

NonaSuomy, 2017-04-28:
Any chance you can copy and paste the origins of the pdf into the README.md, I believe you can even drop pictures on the window. 

The pdf is painful to use, as it's an image of text that you can't search.

Thank you.

NonaSuomy, 2017-04-29:
[PR 79](https://github.com/Edzelf/Esp-radio/pull/79)

Hopefully, this will streamline the process for you.

Edzelf, 2017-04-29:
Wouldn't be too hard to download the pdf file....

NonaSuomy, 2017-04-29:
Not only download it but have to run a GUI and install extra apps also that they become in a stagnate state as even right now the features section of the pdf doesn't match what you have in the readme, pdf's are open source unfriendly and difficult to manage in an ever changing project. Documentation should flow as easy as the source itself or other wise the documentation won't get updated. Never seen any project prefer something like that. You can always export the markdown to a pdf if you require that for some reason.

NonaSuomy, 2017-04-29:
Here's a link to the updated [README.md](https://github.com/NonaSuomy/Esp-radio/blob/master/README.md)

NonaSuomy, 2017-04-29:
If there is anything you want me to fix before accepting the pull request let me know and I'll see what I can do.